GrKoR/esphome_aux_ac_component #2: шаг регулировки и отображения температуры

necroave, 2021-07-21:
В поддерживаемые кондиционеры смело можно записывать Centek CT-65Q09
Потребовалась распайка порта. На самом кондиционере не распаян разъем. CN201.
Распиновка аналогична кондиционеру [Roda RS-AL09F](https://sprut.ai/static/media/cache/00/31/00/40/2151/69356/3000x_image.jpeg)

Есть небольшой нюанс связанный с регулировкой температуры:
#define AC_TEMPERATURE_STEP 0.1
Этот параметр привязан к дальнейшему 
 // the step with which to increase/decrease target temperature. This also affects with how many decimal places the temperature is shown.
 traits.set_visual_temperature_step(AC_TEMPERATURE_STEP);

Похоже что тянется отсюда:  ESPHOME climate_ir.cpp [https://www.esphome.io/api/climate__ir_8cpp_source.html](https://www.esphome.io/api/climate__ir_8cpp_source.html)

При попытке изменить температуру с шагом в 0.1 или 0.5 кондиционер команду не принимает
Команда отправляется, но температура с дробной частью выставляется как целое число с округлением до целого числа в меньшую сторону.
Как пример - была попытка выставить 19.1 и 19.5 градусов:
[09:02:41][D][climate:369]: 'Bedroom AC' - Sending state:
[09:02:41][D][climate:372]:   Mode: OFF
[09:02:41][D][climate:374]:   Action: OFF
[09:02:41][D][climate:377]:   Fan Mode: AUTO
[09:02:41][D][climate:389]:   Swing Mode: OFF
[09:02:41][D][climate:392]:   Current Temperature: 24.70°C
[09:02:41][D][climate:398]:   Target Temperature: 19.00°C

Если исправить #define AC_TEMPERATURE_STEP 0.1 для того чтобы регулировалась температура по 1 градусу и команда проходила корректно - температура с датчика кондиционера начинает отображаться исключительно целыми числами.

Так же с этим параметром связана часть кода в esphome:
`sensor:
  - platform: custom
    lambda: |-
      extern AirCon acAirCon;
      if (!acAirCon.get_initialized()) acAirCon.initAC(id(ac_uart_bus));
      App.register_component(&acAirCon);
      return {acAirCon.sensor_ambient_temperature};
    sensors:
    - name: Bedroom AC ambient temperature
      unit_of_measurement: "°C"
      accuracy_decimals: 1`

accuracy_decimals: 1 в этом случае не влияет на отображение температуры, так как точность в итоге задается через #define AC_TEMPERATURE_STEP 0.1

necroave, 2021-09-03:
Обходное решение:
Использовать кастомизацию в home Assistant:
Добавить в customize.yaml:

```
climate.main_room_ac:
  target_temp_step: 1
```

GrKoR, 2021-09-03:
Centek CT-65Q09 и Roda RS-AL09F внёс в список протестированных.
Шаг изменения температуры буду проверять и исправлять осенью.

GrKoR, 2022-04-06:
поправлено в рамках работы над #10

closed 2022-04-06 by GrKoR