GrKoR/esphome_aux_ac_component #43: возможно неверный контроль пределов целевой температуры

GrKoR, 2022-06-01:
Сейчас минимум и максимум целевой температуры задаются через `Constants::AC_MIN_TEMPERATURE` и `Constants::AC_MAX_TEMPERATURE`. При этом в python проверки на эти пределы нет, можно указать любые.
Brokly сделал в #42 функцию контроля `_temp_target_normalise`, которая не позволяет выходить за пределы диапазона 16..32 градуса.

Но может быть это неверно и стоит дать пользователю возможность задавать меньше/больше? Если он в конфиге esphome задал другие границы, то может он лучше знает своё оборудование?

Можно сделать учет констант `Constants::AC_MIN_TEMPERATURE` и `Constants::AC_MAX_TEMPERATURE` только в случае, если пользователь не задал границы в конфиге. На этапе инициализации компонента заданные пользователем значения или наши константы заносятся в `traits` и в дальнейшем при вызове `_temp_target_normalise` значения берутся уже из `traits` без дополнительных проверок на соответствие `Constants::AC_MIN_TEMPERATURE` и `Constants::AC_MAX_TEMPERATURE`.

Brokly, 2022-06-01:
Как я понял из экспериментов, механизм traits устроен так. То что отдает твой код - esphome::climate::ClimateTraits traits() это некий шаблон, который , после запроса безусловно перегружается другими активными данными, при условии, что они описаны. А то что код запрашивает сам this->get_traits() - возвращает уже консолидированные настройки, окончательные .

GrKoR, 2022-06-01:
Экспериментально установлено, что минимально, куда пускает сплит, - это 15 градусов. Максимально - 33.
Производитель неспроста везде заявляет 16..32 - на этот режим работы рассчитан компрессор.

Похоже, надо ограничивать пользователя диапазоном 16..32. Менять диапазон в этих пределах можно, выходить за него нельзя.
В идеале нужно засунуть проверку в climate.py и выдавать предупреждение еще на этапе сборки исходников из yaml.

GrKoR, 2022-06-10:
Проверил. Контроль пределов целевой температуры верный.
Поэтому issue закрыт.

closed 2022-06-10 by GrKoR