GrKoR/esphome_aux_ac_component #58: Suspected WiFi module not unlocked?

qsxzaqq, 2022-07-19:
The WiFi LED of the air conditioner is off, but the module can receive ping/pong.

```
[DEBUG] 0003721794: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[VERBOSE] Parser: ping packet received
[VERBOSE] Parser: generated ping answer. Waiting for sending.
[VERBOSE] State changed to ACSM_SENDING_PACKET.
[VERBOSE] Sender: sending packet.
[DEBUG] 0003721813: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[VERBOSE] Sender: 18 bytes sent (51 ms).
[VERBOSE] State changed to ACSM_IDLE.
[VERBOSE] State changed to ACSM_RECEIVING_PACKET.
[VERBOSE] Header loaded: timestamp = 0003724771, start byte = BB, packet type = 01, body size = 00
[VERBOSE] Packet loaded: timestamp = 0003724771, start byte = BB, packet type = 01, body size = 00, crc = [43, FF].
[VERBOSE] Loaded 10 bytes for a 8 ms.
[VERBOSE] 0003724771: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[VERBOSE] State changed to ACSM_PARSING_PACKET.
[DEBUG] 0003724771: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[VERBOSE] Parser: ping packet received
[VERBOSE] Parser: generated ping answer. Waiting for sending.
[VERBOSE] State changed to ACSM_SENDING_PACKET.
[VERBOSE] Sender: sending packet.
[DEBUG] 0003724789: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[VERBOSE] Sender: 18 bytes sent (51 ms).
[VERBOSE] State changed to ACSM_IDLE.
[VERBOSE] getStatusSmall: loaded to sequence
[VERBOSE] getStatusBig: loaded to sequence
[VERBOSE] getStatusBigAndSmall: loaded to sequence
[VERBOSE] Sequence [step 0]: step started
[VERBOSE] Sequence [step 0]: small status request generated:
[VERBOSE] 0003724974: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[VERBOSE] State changed to ACSM_SENDING_PACKET.
[VERBOSE] Sender: sending packet.
[DEBUG] 0003724974: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[VERBOSE] Sender: 12 bytes sent (38 ms).
[VERBOSE] State changed to ACSM_IDLE.
[VERBOSE] Sequence [step 1]: step started
[WARN] Sequence  [step 1]: step timed out (it took 580 ms instead of 580 ms)
[VERBOSE] State changed to ACSM_RECEIVING_PACKET.
[VERBOSE] Header loaded: timestamp = 0003727749, start byte = BB, packet type = 01, body size = 00
[VERBOSE] Packet loaded: timestamp = 0003727749, start byte = BB, packet type = 01, body size = 00, crc = [43, FF].
[VERBOSE] Loaded 10 bytes for a 4 ms.
[VERBOSE] 0003727749: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[VERBOSE] State changed to ACSM_PARSING_PACKET.
[DEBUG] 0003727749: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[VERBOSE] Parser: ping packet received
[VERBOSE] Parser: generated ping answer. Waiting for sending.
[VERBOSE] State changed to ACSM_SENDING_PACKET.
[VERBOSE] Sender: sending packet.
[DEBUG] 0003727758: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[VERBOSE] Sender: 18 bytes sent (48 ms).
[VERBOSE] State changed to ACSM_IDLE.

```

GrKoR, 2022-07-19:
Do you can control your air conditioner with esphome?
I mean is it work properly and the only trouble is wifi LED?

P.S. What is the model of your AC?

GrKoR, 2022-07-19:
Google knows several manufacturers for KFR-35GW: Chigo, Gree, Xiaomi, Haier. Which is yours?

When you switch off VERBOSE logger level (set it to DEBUG) is there any other packets in the log or just ping only?

Upload plz log with logger level DEBUG and where you are try to set some mode on the AC

qsxzaqq, 2022-07-19:
```
[DEBUG] 0000023233: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[DEBUG] 0000026206: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[DEBUG] 0000026210: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[DEBUG] 0000028170: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[WARN] Sequence  [step 1]: step timed out (it took 580 ms instead of 580 ms)
[DEBUG] 0000029183: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[DEBUG] 0000029187: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[DEBUG] 0000032160: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[DEBUG] 0000032164: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[DEBUG] 0000035137: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[DEBUG] 0000035141: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[DEBUG] 0000035187: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[WARN] Sequence  [step 1]: step timed out (it took 580 ms instead of 580 ms)
[DEBUG] 0000038115: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[DEBUG] 0000038119: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[DEBUG] 0000041092: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[DEBUG] 0000041096: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[DEBUG] 0000042188: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[WARN] Sequence  [step 1]: step timed out (it took 580 ms instead of 580 ms)
[DEBUG] 0000043077: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[DEBUG] 0000043081: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
```

The aux model of the Chinese version is very strange. They confuse me.

GrKoR, 2022-07-19:
Yep, it is very strange.
May be it has new protocol version or something like this...
If you have an original AUX wifi-module for your AC and you will capture the logs, so I would probably be able to update the component to support your AC...

GrKoR, 2022-07-20:
We need logs of communication between original wifi-module and this type of air conditioners,
I don't know if this type of air conditioners is local to China or if it is a new version for the global market too. In any case, this information will greatly help the development of the aux_ac component.

makkie2002, 2023-01-06:
> Hi I have exactly the same issue. I am in Europe and have AUX-18QC model. Only ping commands work. I want to log serial communication from original module. How can I do that? Can I connect rx to esp8266 in parallel to normal wifi module?

Hi again. The issue was due to my fault :). I forgot to connect the GND of the ESP to the PCB of the AC. Like that the ESP did not have the correct voltage-reference for the tx/rx signals. After connection everything worked fine!

Special thanks to @GrKoR for helping!

Makkie

GrKoR, 2023-01-07:
@qsxzaqq Hello!
I forgot to ask you about GND connection. Did you double-check it for KFR-35GW? @makkie2002 had the very similar problem. And it was solved by reliable gnd connection.

GrKoR, 2023-03-31:
Many details about AUX Aegean Sea ( 爱琴海 ) connection here: #71

closed 2023-03-31 by GrKoR