GrKoR/esphome_aux_ac_component #60: Updating the outdoor temperature even if the air conditioner is turned off

WiZaR5789, 2022-07-22:
I don't know if this topic has already been discussed, but is it possible to update the outside temperature when the air conditioner is turned off?  In the esp logs I can see that communication between the unit and esp is present even when the air conditioner is turned off, however the temperature is not updated.
[image]

GrKoR, 2022-07-23:
I checked the source code superficially and didn't found anything suspicious. The outdoor temperature should be published every time the sensor value has changed.
My air conditioner doesn't have an outdoor temperature sensor, so I can't test this functionality.
If you capture an `aux_ac` log, then I would find the reason for this behavior. Maybe)

The log is needed for the period when the air conditioner is working and showing the temperature, then it is turning off (the temperature is not updated), then it is turning on again (and the temperature is beginning to update).

The better way to capture the log is to use this script: https://github.com/GrKoR/ac_python_logger

WiZaR5789, 2022-07-23:
I am attaching a 5-minute log of operation and non-operation of the air conditioner as well as a photo of the entire outdoor temperature chart. log comes directly from esphome because I had trouble firing your custom logger. It seems that the esp itself logs this value and the air conditioner electronics do not get along with the sensor when it's turned off.

[logs_aux-ac_ON.txt](https://github.com/GrKoR/esphome_aux_ac_component/files/9173916/logs_aux-ac_ON.txt)
[logs_aux-ac_OFF.txt](https://github.com/GrKoR/esphome_aux_ac_component/files/9173944/logs_aux-ac_OFF.txt)

[image]

GrKoR, 2022-07-23:
Great! These logs are also suitable.
I'll be able to take apart to this issue in a couple of weeks, maybe a little earlier.
Thanks for your help!

WiZaR5789, 2022-07-25:
Another observation... It seems that the air conditioner updates this value, however, very rarely every 5-6 hours as you can see in the picture, the temperature gap is huge. The air conditioner has not been turned on for the last 14 hours.

[image]

GrKoR, 2022-07-25:
Very strange. I have no idea why it works so strange. I'll delve into it

GrKoR, 2022-08-24:
@WiZaR5789, I requested additional logs from the community and found that this is not a bug, but a feature =)

This behavior of outdoor temperature is a hardware limitation. When an AC works it often updates outdoor temperature. But when it is off the AC's hardware updates outdoor temperature one time per 6-7 hours. There is no way to get outdoor temperature more often from AC in this mode.

If you need some temperature data from AC I can suggest indirect way. There are some other T-sensors in the outdoor unit: inbound and outbound temperature. These sensors are updated a little bit more often in the power-off AC mode. When AC is works these sensors' temperature are significantly different from the air temperature. One-two minutes later after switching AC off these temperatures are getting closer to the air temperature (but it always differs for 1-2 degrees). You can use dinamics of one of these temperatures for manual change of outdoor temperature when AC is in off mode.

closed 2022-08-24 by GrKoR

GrKoR, 2022-08-24:
I've made changes to the readme

WiZaR5789, 2022-08-24:
Thanks for the investigation, as always, the producer's fault came out, but we won't do anything about it. Again, thank you for your help.

GrKoR, 2022-08-24:
> as always, the producer's fault came out

I think this isn't the fault. The manufacturer probably couldn't even assume that someone would dig deep in the device and use its internal sensors for something.

WiZaR5789, 2022-08-24:
But this could be useful for some automations. If sensor exists and can be used continuously why limit it artificially…

GrKoR, 2022-08-24:
Nobody knows :)