GrKoR/esphome_aux_ac_component #64: Все термодатчики , при выключенном кондее обновляются очень редко :(

Brokly, 2022-09-11:
Привет ! У меня все термометры кондея, кроме внутренних, в выключенном состоянии, обновляются очень редко.
На графике есть два доп. термометра установленные на радиаторе и компрессоре.  
[image]
Никакая интерполяция не спасет :( Кондей не термометр, ни разу :(

GrKoR, 2022-09-11:
Привет!
Да, есть такая трабла. При выключенном кондее апдейт термодатчиков происходит с периодичностью примерно 6.5 часов. Ребята об этом уже репортили ( #60 ). Я проверял по логам - у всех моделей так.
Это фиаско, конечно 😒 
В ридми информацию я внёс. Остаётся как-то с этим жить.

closed 2022-09-11 by GrKoR

Brokly, 2022-09-12:
Не, ну если очень хочется температуру, то нужно попробовать например "выключать" кондей раз в 10 минут. Возможно это заставит его обновить температуру.

GrKoR, 2022-09-12:
Представляю количество проклятий из-за пикающего раз в 10 минут биппера :)))