GrKoR/esphome_aux_ac_component #68: Temperature setpoint setting does not implement 0.5C

makkie2002, 2023-02-01:
Hi,

It Sems that from Home Assistant the temperature setpoint sent to the Airco does not implement the 0.5C. If I send 20.5 it is set to 20.0 …

GrKoR, 2023-02-01:
Ok, I'll check it.
Can you please publish a part of the log where you set temperature and where after 7 seconds AC reports target temperature.

makkie2002, 2023-02-02:
The log file is attached ...
[logs_aux-airco-living_logs.txt](https://github.com/GrKoR/esphome_aux_ac_component/files/10568661/logs_aux-airco-living_logs.txt)

GrKoR, 2023-02-02:
Yes, this is a bug. Will fix it soon.

Also, thanks to your log, I found that the component sends a power limit value when the power limit is off. It doesn't affect functionality, but it is ugly. =)

GrKoR, 2023-02-03:
Hi, @makkie2002 !
I think I fixed it. Check please firmware from v.0.2.10-dev branch.

Use it like this:
```yaml
external_components:
  - source: github://GrKoR/esphome_aux_ac_component@v.0.2.10-dev
    components: [ aux_ac ]
    refresh: 0s
```

And if it is possible publish the part of the log once more. I'll check if the power limitation works like it should.

makkie2002, 2023-02-03:
> Hi, @makkie2002 ! I think I fixed it. Check please firmware from v.0.2.10-dev branch.
>
> Use it like this:
>
> ```yaml
> external_components:
>   - source: [...]

Hi,

Now there is introduced a bigger bug I think :).
- The AC is reporting back a setpoint of 8.0C when the previously send setpoint is 20,0C
- 8.5C is reported when 20.5C has been send
- 9.0C is reported when 21C has been send

Log-file attached: [logs_aux-airco-living_logs (1).txt](https://github.com/GrKoR/esphome_aux_ac_component/files/10583205/logs_aux-airco-living_logs.1.txt)

GrKoR, 2023-02-03:
Yes. You are right.
Stupid bug)
I haven't aux aircons now and as a result can't debug component properly.
I changed bool to uint8. Now it seems should work =)

GrKoR, 2023-02-03:
Thank you for the log! I see that power limit works as it should now.

makkie2002, 2023-02-03:
> Thank you for the log! I see that power limit works as it should now.

Tested and working now. Thanks!

closed 2023-02-03 by makkie2002