GrKoR/esphome_aux_ac_component #70: AUX FREEDOM AUX-07FH/I - compatibility report and unknown packets

diabl0, 2023-03-05:
Multisplit instalation:
external AUX-M3-21LCLH
Internal: two AUX FREEDOM AUX-07FH/I and AUX J-SMART AUX-09JO/I

I had time only for one of Freedom units, and it mainly works. 
Original Wifi module:  CTT-WIFI-BL3353 2022-04-06

Some photos of original module insides:

[image]

[image]

[image]

There is +5V on the white wire (according to my multimeter) but for some reason, the ESP32 dev board won't start - I had to downstep 12V -> 5V from the yellow wire.

Enough space to fit even a large esp dev board (not to mention Wemos or similar mini modules).

I miss horizontal swing settings and occasionally I had this yellow warnings

```
[12:57:04][D][sensor:127]: 'Bedroom AC Indoor Temperature': Sending state 21.00000 °C with 1 decimals of accuracy
[12:57:04][D][sensor:127]: 'Bedroom AC Outdoor Temperature': Sending state 5.00000 °C with 0 decimals of accuracy
[12:57:04][D][sensor:127]: 'Bedroom AC Colant Inbound Temperature': Sending state 16.00000 °C with 0 decimals of accuracy
[12:57:04][D][sensor:127]: 'Bedroom AC Colant Outbound Temperature': Sending state 12.00000 °C with 0 decimals of accuracy
[12:57:04][D][sensor:127]: 'Bedroom AC Compressor Temperature': Sending state 42.00000 °C with 0 decimals of accuracy
[12:57:04][D][sensor:127]: 'Bedroom AC Inverter Power': Sending state 0.00000 % with 0 decimals of accuracy
[12:57:04][D][sensor:127]: 'Bedroom AC VLouvers State': Sending state 0.00000  with 0 decimals of accuracy
[12:57:04][D][text_sensor:067]: 'Bedroom AC Preset Reporter': Sending state 'Health'
[12:57:04][W][AirCon:1903]: Sequence [step 3]: irrelevant incoming packet
[12:57:04][W][AirCon:1904]: Incoming packet:
[12:57:04][W][AirCon:1905]: 0000042729: [<=] [BB 00 07 00 00 00 19 00] 01 2C E0 9D 00 01 28 35 30 30 30 64 25 2C 4A 2C 00 00 A5 64 00 00 00 00 00 [A5 AD] 
[12:57:04][W][AirCon:1906]: Sequence packet needed: PACKET_TYPE = 07, CMD = 21
[12:57:04][W][AirCon:959]: Sequence  [step 3]: error was occur in step function
[12:57:05][D][number:012]: 'Bedroom AC Vertical Louvers': Sending state 0.000000
[12:57:07][D][AirCon:1213]: 0000045642: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[12:57:07][D][AirCon:1479]: 0000045682: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[12:57:07][D][number:012]: 'Bedroom AC Vertical Louvers': Sending state 0.000000
[12:57:09][D][number:012]: 'Bedroom AC Vertical Louvers': Sending state 0.000000
[12:57:10][D][AirCon:1213]: 0000048638: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[12:57:10][D][AirCon:1479]: 0000048677: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[12:57:10][D][AirCon:1479]: 0000049133: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[12:57:10][D][AirCon:1213]: 0000049337: [<=] [BB 00 07 00 00 00 0F 00] 01 11 60 00 03 A0 00 80 00 00 22 00 10 00 00 [97 CD] 
[12:57:10][D][AirCon:1479]: 0000049395: [=>] [BB 00 06 80 00 00 02 00] 21 01 [1B 7E] 
[12:57:11][D][AirCon:1213]: 0000049614: [<=] [BB 00 07 00 00 00 19 00] 01 21 E0 9D 00 01 27 34 30 30 30 64 25 2C 49 2D 00 00 A5 64 00 00 00 09 00 [A7 AF] 
```

```
[13:09:34][D][text_sensor:067]: 'Bedroom AC Preset Reporter': Sending state 'Health'
[13:09:34][W][AirCon:1846]: Sequence [step 1]: irrelevant incoming packet
[13:09:34][W][AirCon:1847]: Incoming packet:
[13:09:34][W][AirCon:1848]: 0000792533: [<=] [BB 00 07 00 00 00 19 00] 01 2C E0 9D 00 02 5A 33 3B 3B 3B 64 24 24 3C 24 1E 09 BD 28 11 00 00 09 00 [25 DD] 
[13:09:34][W][AirCon:1849]: Sequence packet needed: PACKET_TYPE = 07, CMD = 11
[13:09:34][W][AirCon:959]: Sequence  [step 1]: error was occur in step function
[13:09:35][D][number:012]: 'Bedroom AC Vertical Louvers': Sending state 0.000000
[13:09:36][D][AirCon:1213]: 0000795422: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[13:09:37][D][AirCon:1479]: 0000795465: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[13:09:37][D][number:012]: 'Bedroom AC Vertical Louvers': Sending state 0.000000
[13:09:39][D][number:012]: 'Bedroom AC Vertical Louvers': Sending state 0.000000
[13:09:39][D][AirCon:1213]: 0000798426: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[13:09:40][D][AirCon:1479]: 0000798469: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[13:09:40][D][AirCon:1479]: 0000799197: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[13:09:40][D][AirCon:1213]: 0000799386: [<=] [BB 00 07 00 00 00 0F 00] 01 11 60 00 10 A0 00 80 00 00 22 00 10 00 00 [8A CD] 
[13:09:40][D][AirCon:1479]: 0000799447: [=>] [BB 00 06 80 00 00 02 00] 21 01 [1B 7E] 
[13:09:41][D][AirCon:1213]: 0000799671: [<=] [BB 00 07 00 00 00 19 00] 01 21 E0 9D 00 02 5A 33 3B 3B 3B 64 24 23 3D 24 1E 09 BD 28 11 00 00 09 00 [24 E9] 
[13:09:41][D][climate:387]: 'Bedroom AC' - Sending state:
```

Please let me know how else I can help you improve this component and fully support (at least my) AUX units.

I didn't have time to investigate J-Smart unit, but I saw that it has USB connected module - I don't know how it looks under the electric panel cover.

GrKoR, 2023-03-05:
Thank you for your feedback.

> There is +5V on the white wire (according to my multimeter) but for some reason, the ESP32 dev board won't start - I had to downstep 12V -> 5V from the yellow wire.

It isn't allowed to use +5V line as the power supply of ESP, because it is just digital signal that is going directly from MCU of conditioner motherboard. So there is a risk of burning out the conditioner's MCU port. You should use a 12-15V power line only.

> I miss horizontal swing settings and occasionally I had this yellow warnings

Horizontal swing and warnings in log are not related.
Sporadically (period ~10 min) conditioner reports his state without request from esp. It's the normal operation of AC. And sometimes this report may interfere with esp-AC dataflow. It isn't a big deal and in most cases should not affect control of AC.

> Please let me know how else I can help you improve this component and fully support (at least my) AUX units.
Do all the functions of your AC work? Maybe you have some functions which work from IR-remote but don't with esp? I don't completely understand this. 

> I didn't have time to investigate J-Smart unit, but I saw that it has USB connected module - I don't know how it looks under the electric panel cover.

It should work with aux_ac. The wifi-dongle's USB of AUX conditioners isn't the USB at all. It is just UART with +12-15V power and a USB-like connector. Don't insert normal USB devices into it =)

diabl0, 2023-03-11:
Small update:

AUX J-SMART AUX-09JO/I also works fine.
"USB" pinout:

[image]

1 - +12V
2 - RX 
3 - TX
4 - GND

There are still  some crashes when trying to use power limiting functionality (at least after disabling unneeded sensors watchdog can reset esp without the need to do a power cycle):

```
[11:20:52][D][debug:250]: Reset Reason: Software Watchdog
[11:20:52][D][debug:251]: Reset Info: Fatal exception:4 flag:3 (Software Watchdog) epc1:0x4026d23e epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
```

[image]

Used ESP8266 Wemos D1 mini clone. Waiting for the shipment of ESP32 to see if more power/memory will help and play more with power limiting.

closed 2023-03-30 by GrKoR