GrKoR/esphome_aux_ac_component #71: This is my wifi module, one of which does not work properly

yybl, 2023-03-23:
This is my wifi module，
[image]
[image]
[image]

I have two esps, one of which can read information and operate normally (the only disadvantage is that the original wifi module can view the latest information but cannot operate), but the other one with the same wiring and different esps (d1-mini, esp-01s) cannot read information correctly.

`21:36:12	[D]	[AirCon:1213]	
0001144126: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:12	[D]	[AirCon:1479]	
0001144160: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:14	[D]	[sensor:126]	
'AC WiFi Signal': Sending state -47.00000 dBa with 0 decimals of accuracy
21:36:15	[D]	[AirCon:1213]	
0001147093: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:15	[D]	[AirCon:1479]	
0001147127: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:16	[D]	[AirCon:1479]	
0001148197: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
21:36:16	[W]	[AirCon:949]	
Sequence  [step 1]: step timed out (it took 593 ms instead of 580 ms)
21:36:17	[D]	[AirCon:1213]	
0001150072: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:18	[D]	[AirCon:1479]	
0001150109: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:20	[D]	[AirCon:1213]	
0001153042: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:21	[D]	[AirCon:1479]	
0001153096: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:23	[D]	[AirCon:1479]	
0001155200: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
21:36:23	[W]	[AirCon:949]	
Sequence  [step 1]: step timed out (it took 592 ms instead of 580 ms)
21:36:23	[D]	[AirCon:1213]	
0001156025: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:24	[D]	[AirCon:1479]	
0001156062: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:25	[D]	[AirCon:1213]	
0001158007: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:25	[D]	[AirCon:1479]	
0001158031: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:28	[D]	[AirCon:1213]	
0001160978: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:28	[D]	[AirCon:1479]	
0001161031: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:30	[D]	[AirCon:1479]	
0001162210: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
21:36:30	[W]	[AirCon:949]	
Sequence  [step 1]: step timed out (it took 593 ms instead of 580 ms)
21:36:31	[D]	[AirCon:1213]	
0001163970: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:31	[D]	[AirCon:1479]	
0001164006: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:32	[D]	[number:012]	
'AC Vertical Louvers': Sending state 1.000000
21:36:34	[D]	[AirCon:1213]	
0001166945: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:34	[D]	[AirCon:1479]	
0001166981: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:37	[D]	[AirCon:1479]	
0001169207: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
21:36:37	[W]	[AirCon:949]	
Sequence  [step 1]: step timed out (it took 592 ms instead of 580 ms)
21:36:37	[D]	[AirCon:1213]	
0001169909: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:37	[D]	[AirCon:1479]	
0001169963: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:40	[D]	[AirCon:1213]	
0001172897: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:40	[D]	[AirCon:1479]	
0001172934: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:43	[D]	[AirCon:1213]	
0001175872: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:43	[D]	[AirCon:1479]	
0001175909: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:44	[D]	[sensor:126]	
'AC WiFi Signal': Sending state -46.00000 dBa with 0 decimals of accuracy
21:36:44	[D]	[AirCon:1479]	
0001176220: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
21:36:44	[W]	[AirCon:949]	
Sequence  [step 1]: step timed out (it took 593 ms instead of 580 ms)
21:36:46	[D]	[AirCon:1213]	
0001178846: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:46	[D]	[AirCon:1479]	
0001178883: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:49	[D]	[AirCon:1213]	
0001181817: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:49	[D]	[AirCon:1479]	
0001181854: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:51	[D]	[AirCon:1479]	
0001183231: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
21:36:51	[W]	[AirCon:949]	
Sequence  [step 1]: step timed out (it took 592 ms instead of 580 ms)
21:36:52	[D]	[number:012]	
'AC Vertical Louvers': Sending state 1.000000
21:36:52	[D]	[AirCon:1213]	
0001184791: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:52	[D]	[AirCon:1479]	
0001184815: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:55	[D]	[AirCon:1213]	
0001187770: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
21:36:55	[D]	[AirCon:1479]	
0001187791: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
21:36:57	[D]	[AirCon:1213]	

`

yybl, 2023-03-23:
Please forgive me for my poor English

GrKoR, 2023-03-23:
Hi! Please describe in detail your hardware and connections. Do it for the working and failed one separately.

Questions for every hardware:
1. What esp do you use?
2. What pin for AUX_TX → ESP_RX line do you use?
3. What pin for AUX_RX ← ESP_RX line do you use?
4. Did you connect GND of esp board and AC?
5. How do you perform power connection of the esp?
6. Did you disconnect from AC this big board from the photos? Or may be you just disconnected AC_TX & AC_RX wires?
7. If answer is not for (6) then did you check if there is no other connected chips and smd components on these lines on the board?

Your log looks like the esp receives data. But AC isn't receives esp's signals.
The reasons may be different:
- bad connection somewere on the AC_RX ← ESP_TX line (bad soldering, bad connector connections, using of dupont-wires and so on).
- some extra components on this line (TX & RX pins of NodeMCU often has current limiting resistors, your big board probably also has something like this).

yybl, 2023-03-24:
Perhaps this air conditioner is only sold in China.

The first device, which I directly welded to the tx and rx of the wifi module of AUX, can work normally and control AC, indicating that my wiring is correct.

So I started to disassemble the second air conditioner, which was initially directly welded to the tx and rx of the wifi module, but this time it couldn't work properly and obtain information. Then I directly cut off the circuit board of the AUX and soldered it directly onto the+5V RX and+5V TX of the host. The same error was reported. Restore the WIFI module using AUX, it can work normally.

I'll give you more information after work in the evening.

yybl, 2023-03-24:
My normal working air conditioner is welded here
[image]

The second one was also wired like this and couldn't work properly, so I changed to
[image]
 Added a conversion module
5V to 3.3V IIC UART SPI
[image]
[image]

tgrosz, 2023-03-24:
https://github.com/GrKoR/esphome_aux_ac_component/blob/master/README-EN.md#esphome-aux-air-conditioner-custom-component-aux_ac
maybe that’s what you try. it works for my AC

yybl, 2023-03-24:
AUX_ AC1 is working properly
[image]

Aux-ac2 is abnormal

[image]

yybl, 2023-03-26:
The latest progress is that only the RX of ESP is connected, and the WIFI module of AUX is reserved. It is found that the AC status can be obtained, which means that the TX data transmission of ESP is abnormal?

GrKoR, 2023-03-26:
Yes. I write you the same info earlier:

> bad connection somewere on the AC_RX ← ESP_TX line (bad soldering, bad connector connections, using of dupont-wires and so on).

GrKoR, 2023-03-26:
Try to exclude shift converter (5V to 3.3V IIC UART SPI) from your connection. Maybe it makes sense

yybl, 2023-03-30:
Final solution:

1. The connection is shown in the figure below

2. If the TX transmission is abnormal, it is necessary to cut off the TX line of the aux wifi module, as shown in the marked position below
[image]

GrKoR, 2023-03-30:
Tell me please model name (code) of yours conditioners. I'll update the tested list for help other people with such ACs.

yybl, 2023-03-31:
KFR-26GW/BpR3QYQ1+1
KFR-35GW/BpR3QYQ1+1
KFR-26GW/BpR3QYQ2+2
KFR-35GW/BpR3QYQ2+2
KFR-35GW/BpQYD1+1R3
KFR-35GW/BpQYD2+2R3
KFR-26GW/BpR3QYD1+1
KFR-26GW/BpQYD2+2R3
KFR-35GW/BpQYA1+1R3
KFR-35GW/BpQYA2+2R3
KFR-26GW/BpR3QYA1+1
KFR-26GW/BpQYA2+2R3

This is a series of machines,

It's called the Aegean Sea（爱琴海）

yybl, 2023-03-31:
[image]

GrKoR, 2023-03-31:
Thank you! 
I've added this info into the documentation.

closed 2023-03-31 by GrKoR