GrKoR/esphome_aux_ac_component #74: Дополнительные пресеты/режимы

glebsterx, 2023-05-20:
В данный момент поддерживаются следующие дополнительные функции: health, clean (на моем пульте это iclean), antifungus (Anti-F) и сон. 
На пульте есть еще ELE.H и ECO -- честно признаюсь, мне они не особо и нужны, как минимум потому что у меня не инверторный кондей (eco работать не будет), но возможно кому-то пригодятся.
Прикладываю скрин из инструкции с описанием функций.
[image]
[полная версия инструкции](https://klimakom.ru/upload/iblock/df7/df7d6902f5e769cf370bf8cfdf593ce0.pdf)

Прошу рассматривать данный топик лишь как предложение, но не полноценный фичреквест)

GrKoR, 2023-05-21:
С любыми функциями расклад такой, что просто по названию их не запилить. Нужен лог от того, у кого такие функции есть. Последовательность действий такая:
1. Запускаешь [специально написанную тулзу](https://github.com/GrKoR/ac_python_logger) для сбора логов и включешь кондей.
2. Ждёшь 10+ сек, чтобы все возможные типы пакетов прошли по уарту на esp.
3. Включаешь нужную функцию.
4. Опять ждёшь 10+ сек. Пока ждёшь, записываешь в отдельный текстовик или на бумажку, что сделал.
5. Выключаешь нужную функцию.
6. Опять ждёшь 10+ сек. И записываешь, что сделал.
7. Повторяешь шаги 3..6 для всех других функций, которые хочешь проверить.
8. Останавливаешь запись лога скриптом.

Вместо скрипта на python можно и просто логи из вебморды сохранять копипастом или из командной строки, но там много лишнего шлётся. И легко что-то прозевать. Но в принципе тоже вполне рабочий вариант.

По итогу всех махинаций отправляешь собранный лог и свои заметки (пояснения к логу) сюда или в телегу. По ним можно попробовать расшифровать функционал и тогда можно запилить новую фичу.
Мы так фиксированные положения жалюзи у кондея расшифровали. Вот можно посмотреть в чате: https://t.me/aux_ac/6308 .

GrKoR, 2023-05-21:
P.S. Надо, видимо, это описание тоже в ридми зафигачить... чтобы у желающих было понимание, как действовать

GrKoR, 2023-09-12:
Было сделано в https://github.com/GrKoR/esphome_aux_ac_component/commit/8b17ca705200d23eb46027ea22de557b58b5abf1

closed 2023-09-12 by GrKoR