GrKoR/esphome_aux_ac_component #76: осушение при тихом охлаждении

gleba, 2023-06-11:
Может быть это нормально, но всё равно странно. 

В режиме охлаждения оставив пустым Custom Fan Mode получаем 
```
Mode: COOL
Action: COOLING
```
стоит выбрать тихий режим, получаем осушение
```
Mode: COOL
Action: DRYING
Custom Fan Mode: mute
```

<details><summary>Полный лог</summary>
<p>

```
[13:16:53][D][climate:390]:   Mode: COOL
[13:16:53][D][climate:392]:   Action: COOLING
[13:16:53][D][climate:395]:   Fan Mode: LOW
[13:16:53][D][climate:398]:   Custom Fan Mode:
[13:16:53][D][climate:401]:   Preset: NONE
[13:16:53][D][climate:404]:   Custom Preset: 
[13:16:53][D][climate:407]:   Swing Mode: OFF
[13:16:53][D][climate:410]:   Current Temperature: 24.00°C
[13:16:53][D][climate:416]:   Target Temperature: 24.00°C
[13:16:53][D][sensor:127]: 'temp air indoor': Sending state 24.00000 °C with 1 decimals of accuracy
[13:16:53][D][sensor:127]: 'temp air outdoor': Sending state 28.00000 °C with 0 decimals of accuracy
[13:16:53][D][sensor:127]: 'temp inbound': Sending state 13.00000 °C with 0 decimals of accuracy
[13:16:53][D][sensor:127]: 'temp outbound': Sending state 34.00000 °C with 0 decimals of accuracy
[13:16:53][D][sensor:127]: 'temp strange': Sending state 35.00000 °C with 0 decimals of accuracy
[13:16:53][D][AirCon:1213]: 0001781791: [<=] [BB 00 07 00 00 00 04 00] 01 01 F6 DC [42 21] 
[13:16:53][D][AirCon:1479]: 0001781849: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[13:16:54][D][AirCon:1213]: 0001782051: [<=] [BB 00 07 00 00 00 0F 00] 01 11 87 20 00 60 80 20 00 00 20 00 10 00 00 [F6 4C] 
[13:16:54][D][climate:387]: 'Coud Fusion' - Sending state:
[13:16:54][D][climate:390]:   Mode: COOL
[13:16:54][D][climate:392]:   Action: COOLING
[13:16:54][D][climate:395]:   Fan Mode: LOW
[13:16:54][D][climate:398]:   Custom Fan Mode: mute
[13:16:54][D][climate:401]:   Preset: NONE
[13:16:54][D][climate:404]:   Custom Preset: 
[13:16:54][D][climate:407]:   Swing Mode: OFF
[13:16:54][D][climate:410]:   Current Temperature: 24.00°C
[13:16:54][D][climate:416]:   Target Temperature: 24.00°C
[13:16:54][D][sensor:127]: 'temp air indoor': Sending state 24.00000 °C with 1 decimals of accuracy
[13:16:54][D][sensor:127]: 'temp air outdoor': Sending state 28.00000 °C with 0 decimals of accuracy
[13:16:54][D][sensor:127]: 'temp inbound': Sending state 13.00000 °C with 0 decimals of accuracy
[13:16:54][D][sensor:127]: 'temp outbound': Sending state 34.00000 °C with 0 decimals of accuracy
[13:16:54][D][sensor:127]: 'temp strange': Sending state 35.00000 °C with 0 decimals of accuracy
[13:16:55][D][number:012]: 'vlouver state set 0-up  6-down': Sending state 1.000000
[13:16:56][D][AirCon:1213]: 0001785001: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[13:16:57][D][AirCon:1479]: 0001785042: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[13:16:57][D][number:012]: 'vlouver state set 0-up  6-down': Sending state 1.000000
[13:16:59][D][number:012]: 'vlouver state set 0-up  6-down': Sending state 1.000000
[13:16:59][D][AirCon:1479]: 0001787602: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[13:16:59][D][AirCon:1213]: 0001787804: [<=] [BB 00 07 00 00 00 0F 00] 01 11 87 20 00 60 80 20 00 00 20 00 10 00 00 [F6 4C] 
[13:16:59][D][AirCon:1479]: 0001787862: [=>] [BB 00 06 80 00 00 02 00] 21 01 [1B 7E] 
[13:17:00][D][AirCon:1213]: 0001788081: [<=] [BB 00 07 00 00 00 18 00] 01 21 E0 21 00 01 3B 38 2D 2D 2D 64 3C 43 44 39 21 0B 9C E8 15 00 00 00 [5B 81] 
[13:17:00][D][climate:387]: 'Coud Fusion' - Sending state:
[13:17:00][D][climate:390]:   Mode: COOL
[13:17:00][D][climate:392]:   Action: DRYING
[13:17:00][D][climate:395]:   Fan Mode: LOW
[13:17:00][D][climate:398]:   Custom Fan Mode: mute
```

</p>
</details>

gleba, 2023-06-12:
Похоже описанное выше - это штатная работа AC.

Добавлю только замечание что на моей модели CENTEK CT-65V24
Custom Fan Mode: mute работает в Mode: COOL исправно, а в Mode: DRY вроде как не только у меня нет возможности менять скорость вращения вентилятора. 
 - тогда этот комментарий 
https://github.com/GrKoR/esphome_aux_ac_component/blob/cdd97b8da30dd34429c7a84d232c1df4c1981a0a/components/aux_ac/aux_ac.h#L536
- либо не верный, либо относится только какой-то определённому ряду модели например без инвертора.

Brokly, 2023-08-21:
У меня энегролюксы включают муть в любом режиме. 

Что же касается режима охлаждения и осушения. Осушение - частный вариант охлаждения. То есть холодный теплообменник  охлаждает близлежащий воздух до конденсации, из него, воды. Отсутствие же быстрого вращения вентилятора не перемешивает воздух  активно. Однако из соседних слоев в уже охлажденный воздух дифундируют молекулы воды, которые опять же конденсируются на теплообменнике. Кроме того, следует учитывать, что любое охлаждение воздуха, автоматически приводит к выпадению конденсата, то есть осушению. Поскольку индикация текущего режима организована в связке с реальностью, любой режим охлаждения начинается с осушения. Равно как и медленное (условно незаметное) охлаждение и есть осушение.

closed 2023-09-12 by GrKoR