GrKoR/esphome_aux_ac_component #78: Compatibility with domoticz

PeJeWe, 2023-07-25:
Is it possible to make it compatible with domoticz?

GrKoR, 2023-07-25:
I don't know how domoticz work.
You can [use MQTT](https://esphome.io/components/mqtt.html) as a "converter", if domoticz can work with mqtt

closed 2023-07-25 by GrKoR

PeJeWe, 2024-01-02:
I'm sorry to bother you again.
I have a CTTM-40X24-WIFI-AKS Wi-Fi module that was connected to my airco, and i can use this via the ACFreedom app, only i want to use domoticz instead of the ACFreedom app. 
I have set it up to use it via MQTT that is working, i cant get a connection to my AC1 (aux) airco.
I have tried you're ac_logger and this is the outcome:
`G:\Downloads>python ac_logger.py -i 192.168.2.85 -p APICode -l common_data.csv
G:\Downloads\ac_logger.py:70: SyntaxWarning: invalid escape sequence '\w'
  parts = re.search("'([\w ]+)': Got Temperature=([-]?\d+\.\d+)°C", isDallasLog.group(1))
G:\Downloads\ac_logger.py:87: SyntaxWarning: invalid escape sequence '\['
  isAirConLog = re.search("\[AirCon:\d+\]: (.+\])", log.message.decode('utf-8'))
G:\Downloads\ac_logger.py:90: SyntaxWarning: invalid escape sequence '\d'
  if re.match("^\d{10}: ", logstring) is None:
G:\Downloads\ac_logger.py:94: SyntaxWarning: invalid escape sequence '\['
  isDallasLog = re.search("\[dallas.sensor:\d+\]: (.+C)", log.message.decode('utf-8'))
noname 192.168.2.85
APIVersion(major=1, minor=9)
airco @ 192.168.2.85: Connection error occurred: Ping response not received after 90.0 seconds`

this is my yaml file:
`external_components:
  - source: github://GrKoR/esphome_aux_ac_component
    components: [ aux_ac ]
    refresh: 0s

esphome:
  name: airco
  platform: ESP32
  board: nodemcu-32s

# don't forget to set your's wifi settings!
wifi:
  ssid: "WIFISSID"
  password: "PASSWORD"
  manual_ip:
    static_ip: StaticIP
    gateway: 192.168.2.254
    subnet: 255.255.255.0
  ap:
    ssid: AUX Hotspot
    password: "PASSWORD"

api:
    password: "APIpassword"
    
captive_portal:
debug:

logger:
    level: DEBUG
    baud_rate: 0

web_server:
  port: 80
  auth:
    username: username
    password: Password

mqtt:
  broker: IPBROKER
  port: 1883
  username: "username"
  password: "password"
  topic_prefix: "domoticz/"
  discovery: true
  discovery_prefix: "homeassistant"

# UART0 configuration for AUX air conditioner communication
uart:
  id: ac_uart_bus
  # ATTENTION! Use GPIO4 (D2) and GPIO5 (D1) as the TX and RX for NodeMCU-like boards!
  tx_pin: GPIO1
[... 53 lines cut ...]
    ov.getresult()
OSError: [WinError 121] De time-outperiode van de semafoor is verlopen

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "G:\Downloads\ac_logger.py", line 34, in main
    await api.connect(login=True)
  File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aioesphomeapi\client.py", line 307, in connect
    await self.start_connection(on_stop)
  File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aioesphomeapi\client.py", line 333, in start_connection
    await self._execute_connection_coro(self._connection.start_connection())
  File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aioesphomeapi\client.py", line 354, in _execute_connection_coro
    await coro
  File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aioesphomeapi\connection.py", line 618, in start_connection
    raise self._wrap_fatal_connection_exception("starting", ex)
  File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aioesphomeapi\connection.py", line 613, in start_connection
    await start_connect_task
  File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aioesphomeapi\connection.py", line 595, in _do_connect
    await self._connect_socket_connect(await self._connect_resolve_host())
  File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aioesphomeapi\connection.py", line 381, in _connect_socket_connect
    raise SocketAPIError(
aioesphomeapi.core.SocketAPIError: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='ipESPHOME', port=6053))]: [WinError 121] De time-outperiode van de semafoor is verlopen`
 
Can you help me with this?