GrKoR/esphome_aux_ac_component #80: Ошибки в логе

arsenal3008, 2023-09-07:
не знаю, что ему не нравится, но периодически сыпет ошибки и предупреждения:

[11:06:51][D][number:012]: 'AC Vertical Louvers': Sending state 1.000000
[11:06:52][E][AirCon:1205]: Parser: packet CRC fail!
[11:06:52][E][AirCon:1206]: 0000890977: [<=] [BB 00 07 00 00 00 0C 01] 21 C0 04 00 00 00 3A 00 39 00 00 00 [00 00] 
[11:06:52][W][AirCon:1907]: Sequence [step 3]: irrelevant incoming packet
[11:06:52][W][AirCon:1908]: Incoming packet:
[11:06:52][W][AirCon:1909]: 0000000000: [<=] 00 00 00 00 00 00 00 00 08 64 5E 
[11:06:52][W][AirCon:1910]: Sequence packet needed: PACKET_TYPE = 07, CMD = 21
[11:06:52][W][AirCon:963]: Sequence  [step 3]: error was occur in step function
[11:06:52][D][number:012]: 'AC Vertical Louvers': Sending state 1.000000
[11:06:53][D][number:012]: 'AC Vertical Louvers': Sending state 1.000000
[11:06:54][D][number:012]: 'AC Vertical Louvers': Sending state 1.000000
[11:06:54][D][AirCon:1126]: Start byte received but there are some unparsed bytes in the buffer:

[11:06:58][D][number:012]: 'AC Vertical Louvers': Sending state 1.000000
[11:06:58][W][component:204]: Component template.number took a long time for an operation (0.06 s).
[11:06:58][W][component:205]: Components should block for at most 20-30ms.
[11:06:58][D][AirCon:1217]: 0000897904: [<=] [BB 00 07 00 00 00 18 00] 01 21 C0 04 00 00 00 3A 00 39 00 00 00 00 00 00 00 00 00 00 00 00 00 08 [64 5E]

GrKoR, 2023-09-12:
По каким-то причинам приходит битый пакет.
Должен быть такой:
```
[BB 00 07 00 00 00 18 00] 01 21 C0 04 00 00 00 3A 00 39 00 00 00 00 00 00 00 00 00 00 00 00 00 08 [64 5E]
```
В первых квадратных скобках заголовок пакета на 8 байт. В конце в квадратных скобках CRC на 2 байта.
Он у тебя правильный есть в логе в самом конце.

А в начале лога приходит битый.
Вначале прилетает испорченный кусок пакета:
```
[BB 00 07 00 00 00 0C 01] 21 C0 04 00 00 00 3A 00 39 00 00 00 [00 00]
```
потом остаток:
```
00 00 00 00 00 00 00 00 08 64 5E
```

Если сравнишь с правильным пакетом, увидишь, что вместо двух последний байт заголовка (0x18 0x00) прилетает один байт 0x0C.
0x18 в заголовке - это длина тела пакета (24 байта). Так как вместо него прилетает 0x0C (12 байт), то компонент берёт из потока только 12 байт (+2 байта на CRC). И пытается с таким пакетом работать. Но CRC понятное дело не сходится. О чем он и говорит.

Все остальные "лишние" байты он просто выплёвывет в лог для инфы.

То есть у тебя явно какие-то траблы на линии. Может помехи какие-то, которые портят данные. Может контакт где-то не очень... Может уровни сигнала на грани фола и не всегда читаются корректно.
В общем, смотри в сторону железа. Скорее всего где-то там беда.

Подобная история была у кого-то в чате. Года два назад. Но я не помню, у кого и решил ли он её в итоге или нет.

GrKoR, 2023-09-12:
А, ну и предупреждения `Component xxxxxx took a long time...` — это после обновы ESPHome поменялось логирование. И теперь многие компоненты такими предупреждениями спамят. В данном случае это проблема компонента `template.number`, я на неё никак повлиять не могу.

GrKoR, 2023-09-12:
Проблемы из #79 имеют такую же причину, как ошибки в первом сообщении тут.

closed 2023-09-12 by GrKoR