GrKoR/esphome_aux_ac_component #81: Horizontal louvers

VavpyDreko, 2023-09-13:
Hi, is it possible to make hlouvers like vlouvers, so I can make changes to horizontal louvers also?
Thanks,
Tomaz

GrKoR, 2023-09-14:
Hi!
Are you asking for a fixed position of the horizontal louvers?
If so, then I need help to implement this feature. I need an example log where horizontal louvers are set in a specified position by IR-remote. If your AC has this function you can help. There is [a detailed instruction in the documentation](https://github.com/GrKoR/esphome_aux_ac_component/blob/master/docs/HOW_TO_FEATURE_REQUEST-EN.md).

VavpyDreko, 2023-09-14:
Hi,
my AC has just swing function, for vertical and horizontal louvers. I can't put them in specific position. But with your wonderful integration I can change vertical louver position without swing it and then stop it. 
Thank you very much,
Tomaz

PeJeWe, 2024-01-06:
> Hello! In this case we can't implement this feature yet. I was tried to brute_force this values but had no success. So we waiting for a man whose AC has this function. We need logs =)

My AC has this function, only my communication is not yet working :-(
when i have it up and running i will share the log with you

GrKoR, 2024-02-24:
It looks like it's impossible.

closed 2024-02-24 by GrKoR