GrKoR/esphome_aux_ac_component #85: Setting IPower Limit Off doesn't work on my AC

Steamerzone, 2023-10-16:
Hello,

Setting IPower Limit Off doesn't work on my AC, I get the following from the logs:

[19:28:12][D][button:010]: 'IPower Limit Off' Pressed.
[19:28:12][D][AirCon:1645]: 0001199868: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[19:28:12][D][AirCon:1359]: 0001200016: [<=] [BB 00 07 00 00 00 0F 00] 01 11 55 20 00 60 80 80 00 00 20 00 10 E4 00 [27 09] 
[19:28:12][D][AirCon:1645]: 0001200106: [=>] [BB 00 06 80 00 00 0F 00] 01 01 55 20 00 60 80 80 00 00 20 00 10 E4 00 [27 99] 
[19:28:12][W][component:204]: Component aux_ac.climate took a long time for an operation (0.06 s).
[19:28:12][W][component:205]: Components should block for at most 20-30ms.
[19:28:12][D][AirCon:1359]: 0001200271: [<=] [BB 00 07 00 00 00 04 00] 01 01 27 99 [11 65] 
[19:28:12][D][AirCon:1645]: 0001200343: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[19:28:12][D][number:012]: 'Vertical Louvers': Sending state 6.000000
[19:28:12][D][AirCon:1359]: 0001200490: [<=] [BB 00 07 00 00 00 0F 00] 01 11 55 20 00 60 80 80 00 00 20 00 10 E4 00 [27 09] 

Setting the IPower Limit to On seems to work as expected.

Greetings,

Emile

GrKoR, 2023-10-23:
Hey, Emile
Could you provide model number of your air conditioner, please?

GrKoR, 2023-10-23:
@Steamerzone , it looks like your code for `IPower Limit off` button is incorrect. Publish your config, please.

Steamerzone, 2023-10-26:
Hello,

The model number is: HG-12000BTU

I'm using the following config:

```
# UART0 configuration for AUX air conditioner communication
uart:
  id: ac_uart_bus
  # ATTENTION! Use GPIO4 (D2) and GPIO5 (D1) as the TX and RX for NodeMCU-like boards!
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 4800
  data_bits: 8
  parity: EVEN
  stop_bits: 1

climate:
  - platform: aux_ac
    name: "Livingroom AC"
    id: aux_id
    uart_id: ac_uart_bus
    period: 7s
    show_action: true
    display_inverted: false
    timeout: 150
    optimistic: true
    indoor_temperature:
      name: AC Indoor Temperature
      id: ac_indoor_temp
      accuracy_decimals: 1
      internal: false
    outdoor_temperature:
      name: AC Outdoor Temperature
      id: ac_outdoor_temp
      internal: false
    outbound_temperature:
      name: AC Coolant Outbound Temperature
      id: ac_outbound_temp
      internal: false
    inbound_temperature:
      name: AC Coolant Inbound Temperature
      id: ac_inbound_temp
      internal: false
    compressor_temperature:
      name: AC Compressor Temperature
      id: ac_strange_temp
      internal: false
    display_state:
      name: AC Display State
      id: ac_display_state
      internal: false
    defrost_state:
      name: AC Defrost State
      id: ac_defrost_state
      internal: false
    inverter_power:
      name: AC Inverter Power
      id: ac_inverter_power
      internal: false
    inverter_power_limit_value:
      name: AC Inverter Power Limit Value
      id: ac_inverter_power_limit_value
      internal: false
    inverter_power_limit_state:
      name: AC Inverter Power Limit State
      id: ac_inverter_power_limit_state
      internal: false
    preset_reporter:
      name: AC Preset Reporter
      id: ac_preset_reporter
      internal: false
    vlouver_state:
      name: AC Vertical Louvers State
      id: ac_vlouver_state
      internal: false
    visual:
      min_temperature: 16
      max_temperature: 32
      temperature_step: 1
    supported_modes:
      - HEAT_COOL
      - COOL
      - HEAT
      - DRY
      - FAN_ONLY
    custom_fan_modes:
      - MUTE
      - TURBO
    supported_presets:
      - SLEEP
    custom_presets:
      - CLEAN
      - HEALTH
      - ANTIFUNGUS
    supported_swing_modes:
      - VERTICAL
      - HORIZONTAL
      - BOTH

#sensor:
  # just wifi signal strength for debug purpose only
#  - platform: wifi_signal
#    name: WiFi Signal
#    update_interval: 30s
#    unit_of_measurement: "dBa"
#    accuracy_decimals: 0

switch:
  - platform: template
    name: Display
    lambda: |-
      if (id(ac_display_state).state) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      - aux_ac.display_on: aux_id
    turn_off_action:
      - aux_ac.display_off: aux_id

button:
  - platform: template
    name: VLouver Stop
    icon: "mdi:circle-small"
    on_press:
      - aux_ac.vlouver_stop: aux_id
    
  - platform: template
    name: VLouver Swing
    icon: "mdi:pan-vertical"
    on_press:
      - aux_ac.vlouver_swing: aux_id
    
  - platform: template
    name: VLouver Top
    icon: "mdi:pan-up"
    on_press:
      - aux_ac.vlouver_top: aux_id
    
  - platform: template
    name: VLouver Middle Above
    icon: "mdi:pan-top-left"
    on_press:
      - aux_ac.vlouver_middle_above: aux_id
    
  - platform: template
    name: VLouver Middle
    icon: "mdi:pan-left"
    on_press:
      - aux_ac.vlouver_middle: aux_id
    
  - platform: template
    name: VLouver Middle Below
    icon: "mdi:pan-bottom-left"
    on_press:
      - aux_ac.vlouver_middle_below: aux_id
    
  - platform: template
    name: VLouver Bottom
    icon: "mdi:pan-down"
    on_press:
      - aux_ac.vlouver_bottom: aux_id

  - platform: template
    name: IPower Limit Off
    icon: "mdi:power-plug-off-outline"
    on_press:
      - aux_ac.power_limit_off: 
          id: aux_id

  - platform: template
    name: IPower Limit On 100%
    icon: "mdi:fraction-one-half"
    on_press:
      - aux_ac.power_limit_on:
          id: aux_id
          limit: 100

  - platform: template
    name: IPower Limit On 75%
    icon: "mdi:fraction-one-half"
    on_press:
      - aux_ac.power_limit_on:
          id: aux_id
          limit: 75

  - platform: template
    name: IPower Limit On 50%
    icon: "mdi:fraction-one-half"
    on_press:
      - aux_ac.power_limit_on:
          id: aux_id
          limit: 50
  
  - platform: template
    name: IPower Limit On 30%
    icon: "mdi:fraction-one-half"
    on_press:
      - aux_ac.power_limit_on:
          id: aux_id
          limit: 30

number:
  - platform: template
    name: Vertical Louvers
    id: aux_id_vlouver
    icon: "mdi:circle-small"
    mode: "slider"
    min_value: 0
    max_value: 6
    step: 1
    update_interval: 10s
    lambda: |-
      return id(ac_vlouver_state).state;
    set_action:
      then:
        - aux_ac.vlouver_set:
            id: aux_id
            position: !lambda "return x;"

  - platform: template
    name: IPower Limit Value
    id: aux_id_ipower_limit_value
    icon: "mdi:battery-unknown"
    mode: "slider"
    min_value: 30
    max_value: 100
    step: 1
    set_action:
      then:
       - lambda: !lambda |-
                 id(aux_id).powerLimitationOnSequence( x );
```

Greetings,
Emile

GrKoR, 2023-10-26:
Ok, thank you!

Try to delete slider 'IPower Limit Value'. In your config it hasn't off state, synchronization with AC and may affect to the commands.

If that doesn't help, send me please this log: press several buttons (on 30%, on 70%, on 100%, off) and capture the log while you are doing this. 'IPower Limit Value' slider should be deleted from config in this test.

Steamerzone, 2023-10-26:
Hello,

I've disabled the IPower Limit Value slider, but that didn't help, below are the logs: 30% 50% 75% 100% and off.

Greetings,
Emile

```
[22:39:46][D][climate:378]: 'Livingroom AC' - Sending state:
[22:39:46][D][climate:381]:   Mode: HEAT
[22:39:46][D][climate:383]:   Action: HEATING
[22:39:46][D][climate:386]:   Fan Mode: AUTO
[22:39:46][D][climate:389]:   Custom Fan Mode: mute
[22:39:46][D][climate:392]:   Preset: NONE
[22:39:46][D][climate:395]:   Custom Preset: 
[22:39:46][D][climate:398]:   Swing Mode: OFF
[22:39:46][D][climate:401]:   Current Temperature: 19.70°C
[22:39:46][D][climate:407]:   Target Temperature: 19.00°C
[22:39:46][D][sensor:093]: 'AC Indoor Temperature': Sending state 19.70000 °C with 1 decimals of accuracy
[22:39:46][D][sensor:093]: 'AC Outdoor Temperature': Sending state 9.00000 °C with 0 decimals of accuracy
[22:39:46][D][sensor:093]: 'AC Coolant Inbound Temperature': Sending state 39.00000 °C with 0 decimals of accuracy
[22:39:46][D][sensor:093]: 'AC Coolant Outbound Temperature': Sending state 6.00000 °C with 0 decimals of accuracy
[22:39:46][D][sensor:093]: 'AC Compressor Temperature': Sending state 50.00000 °C with 0 decimals of accuracy
[22:39:46][D][sensor:093]: 'AC Inverter Power': Sending state 34.00000 % with 0 decimals of accuracy
[22:39:46][D][sensor:093]: 'AC Vertical Louvers State': Sending state 5.00000  with 0 decimals of accuracy
[22:39:46][D][sensor:093]: 'AC Inverter Power Limit Value': Sending state 100.00000 % with 0 decimals of accuracy
[22:39:46][D][text_sensor:064]: 'AC Preset Reporter': Sending state 'NONE'
[22:39:49][D][AirCon:1359]: 0000374967: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
[22:39:49][D][AirCon:1645]: 0000375005: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[22:39:49][D][button:010]: 'IPower Limit On 30%' Pressed.
[22:39:49][D][AirCon:1645]: 0000375087: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[22:39:49][D][AirCon:1359]: 0000375235: [<=] [BB 00 07 00 00 00 0F 00] 01 11 5C 20 02 A0 80 80 00 00 20 00 10 E4 00 [1D C9] 
[22:39:50][D][AirCon:1645]: 0000375324: [=>] [BB 00 06 80 00 00 0F 00] 01 01 5C 20 00 A0 80 80 00 00 20 00 10 9E 00 [20 9F] 
[22:39:50][W][component:214]: Component aux_ac.climate took a long time for an operation (0.06 s).
[22:39:50][W][component:215]: Components should block for at most 20-30ms.
[22:39:50][D][AirCon:1359]: 0000375489: [<=] [BB 00 07 00 00 00 04 00] 01 01 20 9F [18 5F] 
[22:39:50][D][AirCon:1645]: 0000375561: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[22:39:50][D][AirCon:1359]: 0000375709: [<=] [BB 00 07 00 00 00 0F 00] 01 11 5C 20 00 A0 80 80 00 00 20 00 10 9E 00 [20 0F] 
[22:39:50][D][climate:378]: 'Livingroom AC' - Sending state:
[22:39:50][D][climate:381]:   Mode: HEAT
[22:39:50][D][climate:383]:   Action: HEATING
[22:39:50][D][climate:386]:   Fan Mode: AUTO
[22:39:50][D][climate:389]:   Custom Fan Mode: mute
[22:39:50][D][climate:392]:   Preset: NONE
[22:39:50][D][climate:395]:   Custom Preset: 
[22:39:50][D][climate:398]:   Swing Mode: OFF
[22:39:50][D][climate:401]:   Current Temperature: 19.70°C
[22:39:50][D][climate:407]:   Target Temperature: 19.00°C
[22:39:50][D][sensor:093]: 'AC Indoor Temperature': Sending state 19.70000 °C with 1 decimals of accuracy
[22:39:50][D][sensor:093]: 'AC Outdoor Temperature': Sending state 9.00000 °C with 0 decimals of accuracy
[22:39:50][D][sensor:093]: 'AC Coolant Inbound Temperature': Sending state 39.00000 °C with 0 decimals of accuracy
[22:39:50][D][sensor:093]: 'AC Coolant Outbound Temperature': Sending state 6.00000 °C with 0 decimals of accuracy
[22:39:50][D][sensor:093]: 'AC Compressor Temperature': Sending state 50.00000 °C with 0 decimals of accuracy
[22:39:50][D][sensor:093]: 'AC Inverter Power': Sending state 34.00000 % with 0 decimals of accuracy
[22:39:50][D][sensor:093]: 'AC Vertical Louvers State': Sending state 5.00000  with 0 decimals of accuracy
[22:39:50][D][sensor:093]: 'AC Inverter Power Limit Value': Sending state 30.00000 % with 0 decimals of accuracy
[22:39:50][D][text_sensor:064]: 'AC Preset Reporter': Sending state 'NONE'
[22:39:50][D][button:010]: 'IPower Limit On 50%' Pressed.
[22:39:51][D][AirCon:1645]: 0000376287: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[22:39:51][D][AirCon:1359]: 0000376436: [<=] [BB 00 07 00 00 00 0F 00] 01 11 5C 20 00 A0 80 80 00 00 20 00 10 9E 00 [20 0F] 
[22:39:51][D][AirCon:1645]: 0000376526: [=>] [BB 00 06 80 00 00 0F 00] 01 01 5C 20 00 A0 80 80 00 00 20 00 10 B2 00 [20 8B] 
[22:39:51][W][component:214]: Component aux_ac.climate took a long time for an operation (0.06 s).
[22:39:51][W][component:215]: Components should block for at most 20-30ms.
[22:39:51][D][AirCon:1359]: 0000376692: [<=] [BB 00 07 00 00 00 04 00] 01 01 20 8B [18 73] 
[22:39:51][D][AirCon:1645]: 0000376763: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[22:39:51][D][AirCon:1359]: 0000376911: [<=] [BB 00 07 00 00 00 0F 00] 01 11 5C 20 00 A0 80 80 00 00 20 00 10 B2 00 [1F FB] 
[22:39:51][D][climate:378]: 'Livingroom AC' - Sending state:
[22:39:51][D][climate:381]:   Mode: HEAT
[22:39:51][D][climate:383]:   Action: HEATING
[22:39:51][D][climate:386]:   Fan Mode: AUTO
[... 105 lines cut ...]
[22:39:59][D][AirCon:1645]: 0000385119: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
[22:40:00][D][AirCon:1645]: 0000385608: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
[22:40:00][D][AirCon:1359]: 0000385755: [<=] [BB 00 07 00 00 00 0F 00] 01 11 5C 20 01 A0 80 80 00 00 20 00 10 E4 00 [1E C9] 
[... cut ...]
[22:40:00][D][sensor:093]: 'AC Compressor Temperature': Sending state 50.00000 °C with 0 decimals of accuracy
[22:40:00][D][sensor:093]: 'AC Inverter Power': Sending state 34.00000 % with 0 decimals of accuracy
[22:40:00][D][sensor:093]: 'AC Vertical Louvers State': Sending state 5.00000  with 0 decimals of accuracy
[22:40:00][D][sensor:093]: 'AC Inverter Power Limit Value': Sending state 100.00000 % with 0 decimals of accuracy
[22:40:00][D][text_sensor:064]: 'AC Preset Reporter': Sending state 'NONE'
```

GrKoR, 2023-10-28:
@Steamerzone thank you a lot! I found a bug with your help.
Try `dev`-brunch. Is it solve the problem with disabling of power limitation? If it is okay, I'll merge it into `master`.

Steamerzone, 2023-10-29:
Hello,

Yes the dev-branch fixed it, it is working correctly now.
Thanks!

Emile

GrKoR, 2023-10-29:
I merged it into the master.
Thank you for your help! =)

closed 2023-10-29 by GrKoR