GrKoR/esphome_aux_ac_component #93: Help wanted AC1 (sold in NL) Brand AUX

PeJeWe, 2024-01-02:
I'm sorry to bother you again.
I have a CTTM-40X24-WIFI-AKS Wi-Fi module that was connected to my airco, and i can use this via the ACFreedom app, only i want to use domoticz instead of the ACFreedom app.
I have set it up to use it via MQTT that is working, i cant get a connection to my AC1 (aux) airco.
I have tried you're ac_logger and this is the outcome:
G:\Downloads>python ac_logger.py -i IPeSPHOME -p APICode -l common_data.csv G:\Downloads\ac_logger.py:70: SyntaxWarning: invalid escape sequence '\w' parts = re.search("'([\w ]+)': Got Temperature=([-]?\d+\.\d+)°C", isDallasLog.group(1)) G:\Downloads\ac_logger.py:87: SyntaxWarning: invalid escape sequence '\[' isAirConLog = re.search("\[AirCon:\d+\]: (.+\])", log.message.decode('utf-8')) G:\Downloads\ac_logger.py:90: SyntaxWarning: invalid escape sequence '\d' if re.match("^\d{10}: ", logstring) is None: G:\Downloads\ac_logger.py:94: SyntaxWarning: invalid escape sequence '\[' isDallasLog = re.search("\[dallas.sensor:\d+\]: (.+C)", log.message.decode('utf-8')) noname IPeSPHOME APIVersion(major=1, minor=9) airco @ IPeSPHOME: Connection error occurred: Ping response not received after 90.0 seconds

this is my yaml file:
`external_components:

source: github://GrKoR/esphome_aux_ac_component
components: [ aux_ac ]
refresh: 0s
esphome:
name: airco
platform: ESP32
board: nodemcu-32s

don't forget to set your's wifi settings!
wifi:
ssid: "WIFISSID"
password: "PASSWORD"
manual_ip:
static_ip: StaticIP
gateway: 192.168.2.254
subnet: 255.255.255.0
ap:
ssid: AUX Hotspot
password: "PASSWORD"

api:
password: "APIpassword"

captive_portal:
debug:

logger:
level: DEBUG
baud_rate: 0

web_server:
port: 80
auth:
username: username
password: Password

mqtt:
broker: IPBROKER
port: 1883
username: "username"
password: "password"
topic_prefix: "domoticz/"
discovery: true
discovery_prefix: "homeassistant"

UART0 configuration for AUX air conditioner communication
uart:
id: ac_uart_bus

ATTENTION! Use GPIO4 (D2) and GPIO5 (D1) as the TX and RX for NodeMCU-like boards!
tx_pin: GPIO1
rx_pin: GPIO3
baud_rate: 4800
data_bits: 8
parity: EVEN
stop_bits: 1`

usernames, passwords and the IP's are not for real.
I can communicate via the ESPHOME and Domoticz.
only no communication from the ESPHOME to the airco.
when i change the tx/rx i get the following error:

`G:\Downloads>python ac_logger.py -i IPeSPHOME -p APIeSPHOME -l common_data.csv
G:\Downloads\ac_logger.py:70: SyntaxWarning: invalid escape sequence '\w'
parts = re.search("'([\w ]+)': Got Temperature=([-]?\d+.\d+)°C", isDallasLog.group(1))
G:\Downloads\ac_logger.py:87: SyntaxWarning: invalid escape sequence '['
isAirConLog = re.search("[AirCon:\d+]: (.+])", log.message.decode('utf-8'))
G:\Downloads\ac_logger.py:90: SyntaxWarning: invalid escape sequence '\d'
if re.match("^\d{10}: ", logstring) is None:
G:\Downloads\ac_logger.py:94: SyntaxWarning: invalid escape sequence '['
isDallasLog = re.search("[dallas.sensor:\d+]: (.+C)", log.message.decode('utf-8'))
noname ipESPHOME
APIVersion(major=1, minor=9)

G:\Downloads>python ac_logger.py -i ipESPHOME -p APIeSPHOME -l common_data.csv
G:\Downloads\ac_logger.py:70: SyntaxWarning: invalid escape sequence '\w'
parts = re.search("'([\w ]+)': Got Temperature=([-]?\d+.\d+)°C", isDallasLog.group(1))
G:\Downloads\ac_logger.py:87: SyntaxWarning: invalid escape sequence '['
isAirConLog = re.search("[AirCon:\d+]: (.+])", log.message.decode('utf-8'))
G:\Downloads\ac_logger.py:90: SyntaxWarning: invalid escape sequence '\d'
if re.match("^\d{10}: ", logstring) is None:
G:\Downloads\ac_logger.py:94: SyntaxWarning: invalid escape sequence '['
isDallasLog = re.search("[dallas.sensor:\d+]: (.+C)", log.message.decode('utf-8'))
noname ipESPHOME
Task exception was never retrieved
future: <Task finished name='Task-1' coro=<main() done, defined at G:\Downloads\ac_logger.py:29> exception=SocketAPIError("Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='ipESPHOME', port=6053))]: [WinError 121] De time-outperiode van de semafoor is verlopen")>
Traceback (most recent call last):
File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aioesphomeapi\connection.py", line 365, in _connect_socket_connect
sock = await aiohappyeyeballs.start_connection(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aiohappyeyeballs\impl.py", line 91, in start_connection
raise all_exceptions[0]
File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aiohappyeyeballs\impl.py", line 69, in start_connection
sock = await _connect_sock(
^^^^^^^^^^^^^^^^^^^^
File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aiohappyeyeballs\impl.py", line 146, in _connect_sock
await loop.sock_connect(sock, address)
File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\asyncio\proactor_events.py", line 727, in sock_connect
return await self._proactor.connect(sock, address)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\asyncio\windows_events.py", line 798, in _poll
value = callback(transferred, key, ov)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\asyncio\windows_events.py", line 594, in finish_connect
ov.getresult()
OSError: [WinError 121] De time-outperiode van de semafoor is verlopen

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
File "G:\Downloads\ac_logger.py", line 34, in main
await api.connect(login=True)
File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aioesphomeapi\client.py", line 307, in connect
await self.start_connection(on_stop)
File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aioesphomeapi\client.py", line 333, in start_connection
await self._execute_connection_coro(self._connection.start_connection())
File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aioesphomeapi\client.py", line 354, in _execute_connection_coro
await coro
File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aioesphomeapi\connection.py", line 618, in start_connection
raise self._wrap_fatal_connection_exception("starting", ex)
File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aioesphomeapi\connection.py", line 613, in start_connection
await start_connect_task
File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aioesphomeapi\connection.py", line 595, in _do_connect
await self._connect_socket_connect(await self._connect_resolve_host())
File "C:\Users\User\AppData\Local\Programs\Python\Python312\Lib\site-packages\aioesphomeapi\connection.py", line 381, in _connect_socket_connect
raise SocketAPIError(
aioesphomeapi.core.SocketAPIError: Error connecting to [AddrInfo(family=<AddressFamily.AF_INET: 2>, type=<SocketKind.SOCK_STREAM: 1>, proto=6, sockaddr=IPv4Sockaddr(address='ipESPHOME', port=6053))]: [WinError 121] De time-outperiode van de semafoor is verlopen`

Can you help me with this?

PeJeWe, 2024-01-02:
[image]

PeJeWe, 2024-01-04:
I have a CTTM-40X24-WIFI-AKS Wi-Fi module that was connected to my airco, and i can use this with the ACFreedom app,  i  want to use domoticz instead of the ACFreedom app.
I have set it up to use it with MQTT, thats working
I don't have communication from my AC1 (aux) airco.
I have tried you're ac_logger and this is the outcome:

`G:\Downloads>python ac_logger.py -i 192.168.1.** -p APIeSPHOME -l common_data.csv
G:\Downloads\ac_logger.py:70: SyntaxWarning: invalid escape sequence '\w'
parts = re.search("'([\w ]+)': Got Temperature=([-]?\d+.\d+)°C", isDallasLog.group(1))
G:\Downloads\ac_logger.py:87: SyntaxWarning: invalid escape sequence '['
isAirConLog = re.search("[AirCon:\d+]: (.+])", log.message.decode('utf-8'))
G:\Downloads\ac_logger.py:90: SyntaxWarning: invalid escape sequence '\d'
if re.match("^\d{10}: ", logstring) is None:
G:\Downloads\ac_logger.py:94: SyntaxWarning: invalid escape sequence '['
isDallasLog = re.search("[dallas.sensor:\d+]: (.+C)", log.message.decode('utf-8'))
noname 192.168.1.**
APIVersion(major=1, minor=9)
`
When i push a button on my IR remote control, there is no log on screen

this is my yaml file:
[aux_ac.zip](https://github.com/GrKoR/esphome_aux_ac_component/files/13836230/aux_ac.zip)

____________________________________________________________
external_components:

source: github://GrKoR/esphome_aux_ac_component
components: [ aux_ac ]
refresh: 0s
esphome:
name: airco
platform: ESP32
board: nodemcu-32s

don't forget to set your's wifi settings!
wifi:
ssid: "WIFISSID"
password: "PASSWORD"
manual_ip:
static_ip: 192.168.1.**
gateway: 192.168.2.254
subnet: 255.255.255.0
ap:
ssid: AUX Hotspot
password: "PASSWORD"

api:
password: "APIpassword"

captive_portal:
debug:

logger:
level: DEBUG
baud_rate: 0

web_server:
port: 80
auth:
username: username
password: Password

mqtt:
broker: IPBROKER
port: 1883
username: "username"
password: "password"
topic_prefix: "domoticz/"
discovery: true
discovery_prefix: "homeassistant"

UART0 configuration for AUX air conditioner communication
uart:
id: ac_uart_bus

ATTENTION! Use GPIO4 (D2) and GPIO5 (D1) as the TX and RX for NodeMCU-like boards!
tx_pin: GPIO1
rx_pin: GPIO3
baud_rate: 4800
data_bits: 8
parity: EVEN
stop_bits: 1
______________________________________________________________________

_usernames, passwords and the IP's are changed._

The ESPHOME can communicate with mqtt to Domoticz.

There is no communication between the ESPHOME and the airco in the dashboard log and ac_logger

When i change the tx/rx i get the following error:

`G:\Downloads>python ac_logger.py -i 192.168.1.** -p APICode -l common_data.csv G:\Downloads\ac_logger.py:70: SyntaxWarning: invalid escape sequence '\w' parts = re.search("'([\w ]+)': Got Temperature=([-]?\d+\.\d+)°C", isDallasLog.group(1)) G:\Downloads\ac_logger.py:87: SyntaxWarning: invalid escape sequence '\[' isAirConLog = re.search("\[AirCon:\d+\]: (.+\])", log.message.decode('utf-8')) G:\Downloads\ac_logger.py:90: SyntaxWarning: invalid escape sequence '\d' if re.match("^\d{10}: ", logstring) is None: G:\Downloads\ac_logger.py:94: SyntaxWarning: invalid escape sequence '\[' isDallasLog = re.search("\[dallas.sensor:\d+\]: (.+C)", log.message.decode('utf-8')) 
noname 192.168.1.** 
APIVersion(major=1, minor=9) 
airco @ 192.168.1.**: Connection error occurred: Ping response not received after 90.0 seconds`

I hope its more clear now.

GrKoR, 2024-01-05:
It much much better now =)

You forget to implement component in your configuration. You just configured esphome, uart, api, logger and other system components, but didn't configure aux_ac component. At least you should add to your config something like this:
```yaml
climate:
  - platform: aux_ac
    name: "AC Name"
```
Check [this chapter of readme](https://github.com/GrKoR/esphome_aux_ac_component/blob/master/README-EN.md#aux_ac-configuration) for detailed description.

PS
When you added comment to your old issue #78 ("Compatibility with domoticz"), I saw it right away. There was no need to create a new issue.
If you have any troubles or questions in the future, please write here, don't create a new issue. Even if this issue is closed. I close it because it isn't related my component's code.

closed 2024-01-05 by GrKoR

PeJeWe, 2024-01-06:
> It much much better now =)
>
> You forget to implement component in your configuration. You just configured esphome, uart, api, logger and other system components, but didn't configure aux_ac [...]

I have added the climate section.

still no communication with the ac_logger.

when i make a switch in de esphome dashboard i get this in te dashboard log:

`19:43:10 | [D] | [api:102] | Accepted 192.168.1.**
-- | -- | -- | --
19:43:10 | [V] | [api.connection:1103] | Hello from client: 'aioesphomeapi' \| 192.168.1.** \| API Version 1.9
19:43:10 | [D] | [api.connection:1121] | aioesphomeapi (192.168.2.97): Connected successfully
19:43:22 | [D] | [climate:011] | 'aircos' - Setting
19:43:22 | [D] | [climate:015] | Mode: FAN_ONLY
19:43:22 | [E] | [AirCon:3380] | commandSequence: no pings from HVAC. It seems like no AC connected.
19:43:22 | [D] | [climate:396] | 'aircos' - Sending state:
19:43:22 | [D] | [climate:399] | Mode: FAN_ONLY
19:43:22 | [D] | [climate:401] | Action: IDLE
19:43:22 | [D] | [climate:404] | Fan Mode: LOW
19:43:22 | [D] | [climate:410] | Preset: NONE
19:43:22 | [D] | [climate:416] | Swing Mode: OFF
19:43:22 | [D] | [climate:419] | Current Temperature: nan°C
19:43:22 | [D] | [climate:425] | Target Temperature: nan°C
19:43:22 | [V] | [json:038] | Attempting to allocate 512 bytes for JSON serialization
19:43:22 | [V] | [json:058] | Size after shrink 248 bytes
19:43:22 | [V] | [mqtt:474] | Publish(topic='domoticz//climate/aircos/mode/state' payload='fan_only' retain=1)
19:43:22 | [V] | [mqtt:474] | Publish(topic='domoticz//climate/aircos/target_temperature/state' payload='nan' retain=1)
19:43:22 | [V] | [mqtt:474] | Publish(topic='domoticz//climate/aircos/preset/state' payload='' retain=1)
19:43:22 | [V] | [mqtt:474] | Publish(topic='domoticz//climate/aircos/action/state' payload='idle' retain=1)
19:43:22 | [V] | [mqtt:474] | Publish(topic='domoticz//climate/aircos/fan_mode/state' payload='' retain=1)
19:43:22 | [V] | [mqtt:474] | Publish(topic='domoticz//climate/aircos/swing_mode/state' payload='off' retain=1)
19:43:22 | [W] | [component:214] | Component web_server took a long time for an operation (0.08 s).
19:43:22 | [W] | [component:215] | Components should block for at most 20-30ms.`

I have also switched the rx/tx wires, no luck :-(
Is there a way to check if the uart connection is good or bad?

PeJeWe, 2024-01-11:
Could there be other baudrates for other AUX ac's?

GrKoR, 2024-01-11:
> Could there be other baudrates for other AUX ac's?

As I know, no. All known me AUX-based ACs are using the same baudrate.

You can check connection with simple usb-ttl converter and any terminal program on a computer. Don't forget to connect GND of usb-ttl converter with GND of AC

PeJeWe, 2024-01-11:
I have a usb to ttl dongle.

When I select de wright parameters:
4800 baudrate, 8data-bit, 1stop-bit,  party even.
I don't get readable data.
I see squares separated with a "f" that keeps repeating.
The data don't changes when I press a button on my IR remove.
I have test multiple baudrate 2400, 4800, 9600.
When I select 2400 baudrate, I see zero's separated by spaces.
When I select more then 9600, I don't get any results.
What kind of data should I see?

GrKoR, 2024-01-11:
Squares separated with "f" are looking promising.

AC sends ping-request every ~3 sec. and waiting for response.
Ping-request structure:
byte N | NAME | HEX value
-- | -- | --
0 | START | 0xBB
1 | ? | 0x00
2 | TYPE | 0x01
3 | wifi? | 0x00
4 | ? | 0x00
5 | ? | 0x00
6 | LEN | 0x00
7 | ? | 0x00
8 | CRC1 | 0x43
9 | CRC2 | 0xFF

If you configure your terminal to binary mode (show hex values of bytes instead of ascii-symbols) you should see bytes from the table above. Check it out if you still have doubts.

I suppose you have troubles with uart connection between your board and AC. Verify it with a huge attention, please. And check it for compliance with [corresponding part of the readme (step 2)](https://github.com/GrKoR/esphome_aux_ac_component/blob/master/README-EN.md#installing).

GrKoR, 2024-01-11:
Well...
I looked your config yaml once more. You are using 'nodemcu-32s' board.
Have you done what is indicated in [the comment here (uart configuration)](https://github.com/GrKoR/esphome_aux_ac_component/blob/master/README-EN.md#installing) and [here](https://github.com/GrKoR/esphome_aux_ac_component/blob/master/docs/HARDWARE-EN.md#connections)?
```yaml
uart:
  id: ac_uart_bus
  # ATTENTION! For TX and RX use GPIO4 (D2) and GPIO5 (D1) for NodeMCU-like boards!
  # See docs for details: https://github.com/GrKoR/esphome_aux_ac_component/blob/master/docs/HARDWARE-EN.md
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 4800
  data_bits: 8
  parity: EVEN
  stop_bits: 1
```
I mean did you try digital pins instead of uatr tx and rx? The reason of this needs is described [here](https://github.com/GrKoR/esphome_aux_ac_component/blob/master/docs/HARDWARE-EN.md#connections).

PeJeWe, 2024-01-11:
i get this outcome:
BB 00 BB 00 01 00 00 00 00 00 43 FF BB 00 01 00 00 00 00 00 43 FF BB 00 
01 00 00 00 00 00 43 FF BB 00 01 00 00 00 00 00 43 FF BB 00 01 00 00 00 
00 00 43 FF BB 00 01 00 00 00 00 00 43 FF BB 00 01 00 00 00 00 00 43 FF 
BB 00 01 00 00 00 00 00 43 FF BB 00 01 00 00 00 00 00 43 FF BB 00 01 00 
00 00 00 00 43 FF BB 00 01 00 00 00 00 00 43 FF BB 00 01 00 00 00 00 00 
43 FF BB 00 01 00 00 00 00 00 43 FF BB 00 01 00 00 00 00 00 43 FF BB 00 
01 00 00 00 00 00 43 FF

PeJeWe, 2024-01-11:
Got it working by changing the GPIO ports.

noname;2024-01-11 21:53:37;0000139806;[=>];BB;00;01;80;01;00;08;00;1C;27;00;00;00;00;00;00;1E;58
ping skipped

noname;2024-01-11 21:53:37;0000140278;[=>];BB;00;06;80;00;00;02;00;11;01;2B;7E

noname;2024-01-11 21:53:37;0000140490;[<=];BB;00;07;00;00;00;0F;00;01;11;85;20;01;60;00;C0;00;00;20;00;10;00;00;76;AD

noname;2024-01-11 21:53:37;0000140591;[=>];BB;00;06;80;00;00;02;00;21;01;1B;7E

noname;2024-01-11 21:53:37;0000140841;[<=];BB;00;07;00;00;00;18;00;01;21;E0;C1;00;02;55;32;33;33;33;64;24;23;27;39;00;00;9A;04;00;00;00;04;A2;EB

noname;2024-01-11 21:53:40;0000143745;[<=];BB;00;01;00;00;00;00;00;43;FF
ping skipped

noname;2024-01-11 21:53:40;0000143817;[=>];BB;00;01;80;01;00;08;00;1C;27;00;00;00;00;00;00;1E;58
ping skipped

noname;2024-01-11 21:53:44;0000146725;[<=];BB;00;01;00;00;00;00;00;43;FF
ping skipped

noname;2024-01-11 21:53:44;0000146789;[=>];BB;00;01;80;01;00;08;00;1C;27;00;00;00;00;00;00;1E;58
ping skipped

noname;2024-01-11 21:53:44;0000147287;[=>];BB;00;06;80;00;00;02;00;11;01;2B;7E

noname;2024-01-11 21:53:44;0000147504;[<=];BB;00;07;00;00;00;0F;00;01;11;85;20;01;60;00;C0;00;00;20;00;10;00;00;76;AD

noname;2024-01-11 21:53:44;0000147604;[=>];BB;00;06;80;00;00;02;00;21;01;1B;7E

noname;2024-01-11 21:53:44;0000147847;[<=];BB;00;07;00;00;00;18;00;01;21;E0;C1;00;02;54;32;33;33;33;64;24;23;27;39;00;00;9A;04;00;00;00;04;A3;EB

noname;2024-01-11 21:53:48;0000150685;[=>];BB;00;06;80;00;00;02;00;11;01;2B;7E

noname;2024-01-11 21:53:48;0000150893;[<=];BB;00;07;00;00;00;0F;00;01;11;85;20;01;60;00;C0;00;00;20;00;10;00;00;76;AD

noname;2024-01-11 21:53:48;0000150993;[=>];BB;00;06;80;00;00;0F;00;01;01;85;20;00;60;00;C0;00;00;00;00;10;1E;00;98;1F

noname;2024-01-11 21:53:48;0000151213;[<=];BB;00;07;00;00;00;04;00;01;01;98;1F;A0;DE

noname;2024-01-11 21:53:48;0000151316;[=>];BB;00;06;80;00;00;02;00;11;01;2B;7E

noname;2024-01-11 21:53:48;0000151542;[<=];BB;00;07;00;00;00;0F;00;01;11;85;20;00;60;00;C0;00;00;00;00;10;1E;00;97;8F

noname;2024-01-11 21:53:51;0000154302;[=>];BB;00;06;80;00;00;02;00;11;01;2B;7E

noname;2024-01-11 21:53:51;0000154515;[<=];BB;00;07;00;00;00;0F;00;01;11;85;20;00;60;00;C0;00;00;00;00;10;00;00;97;AD

noname;2024-01-11 21:53:51;0000154754;[=>];BB;00;06;80;00;00;02;00;21;01;1B;7E

noname;2024-01-11 21:53:51;0000154983;[<=];BB;00;07;00;00;00;18;00;01;21;E0;C0;00;00;00;32;32;32;32;64;24;23;27;39;00;00;9A;04;00;00;00;04;F9;EF

noname;2024-01-11 21:53:54;0000157905;[<=];BB;00;01;00;00;00;00;00;43;FF
ping skipped

noname;2024-01-11 21:53:54;0000157960;[=>];BB;00;01;80;01;00;08;00;1C;27;00;00;00;00;00;00;1E;58
ping skipped

noname;2024-01-11 21:53:58;0000160888;[<=];BB;00;01;00;00;00;00;00;43;FF
ping skipped

noname;2024-01-11 21:53:58;0000160953;[=>];BB;00;01;80;01;00;08;00;1C;27;00;00;00;00;00;00;1E;58
ping skipped

noname;2024-01-11 21:53:58;0000161315;[=>];BB;00;06;80;00;00;02;00;11;01;2B;7E

noname;2024-01-11 21:53:58;0000161524;[<=];BB;00;07;00;00;00;0F;00;01;11;85;20;00;60;00;C0;00;00;00;00;10;00;00;97;AD

noname;2024-01-11 21:53:58;0000161625;[=>];BB;00;06;80;00;00;02;00;21;01;1B;7E

noname;2024-01-11 21:53:58;0000161853;[<=];BB;00;07;00;00;00;18;00;01;21;E0;C0;00;00;00;32;32;32;32;64;24;23;27;39;00;00;9A;04;00;00;00;04;F9;EF

noname;2024-01-11 21:54:01;0000164789;[<=];BB;00;01;00;00;00;00;00;43;FF
ping skipped

noname;2024-01-11 21:54:01;0000164859;[=>];BB;00;01;80;01;00;08;00;1C;27;00;00;00;00;00;00;1E;58
ping skipped

noname;2024-01-11 21:54:05;0000167757;[<=];BB;00;01;00;00;00;00;00;43;FF
ping skipped

noname;2024-01-11 21:54:05;0000167827;[=>];BB;00;01;80;01;00;08;00;1C;27;00;00;00;00;00;00;1E;58
ping skipped

noname;2024-01-11 21:54:05;0000168325;[=>];BB;00;06;80;00;00;02;00;11;01;2B;7E

noname;2024-01-11 21:54:05;0000168534;[<=];BB;00;07;00;00;00;0F;00;01;11;85;20;01;60;00;C0;00;00;00;00;10;00;00;96;AD

noname;2024-01-11 21:54:05;0000168633;[=>];BB;00;06;80;00;00;02;00;21;01;1B;7E

noname;2024-01-11 21:54:05;0000168879;[<=];BB;00;07;00;00;00;18;00;01;21;E0;C0;00;00;00;32;32;32;32;64;24;23;27;39;00;00;9A;04;00;00;00;03;F9;F0

noname;2024-01-11 21:54:08;0000171790;[<=];BB;00;01;00;00;00;00;00;43;FF
ping skipped

noname;2024-01-11 21:54:08;0000171858;[=>];BB;00;01;80;01;00;08;00;1C;27;00;00;00;00;00;00;1E;58

GrKoR, 2024-01-11:
Cool. Now component is working.

PeJeWe, 2024-01-11:
What is the best way to find te setting that sets the louver in de down state.
I can set my louver in any direction in 6 steps, and the last one is swing up/down automatically.

GrKoR, 2024-01-11:
The vertical movement of the louvers is already decoded (there are actions for vertical louvers in the component). The only thing around louvers I am looking for is horizontal movement with fixed position. Something like centre/left/right. But I don't know if it's really possible for AUX.
Do your AC maintain this feature?

PeJeWe, 2024-01-12:
For the horizontal movement my ac does not have that function.
I only can stop them at certain places.
Ok, thanks for the great help

GrKoR, 2024-01-12:
You are welcome 🤝

PeJeWe, 2024-01-13:
I had yesterday good communication, and I could set it on and off and other stuff. 
I can not today.
`Time	level	Tag	Message
10:33:06	[D]	[AirCon:1367]	
0000229944: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
10:33:06	[D]	[AirCon:1651]	
0000229991: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
10:33:07	[D]	[AirCon:1651]	
0000231373: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
10:33:08	[W]	[AirCon:1067]	
Sequence  [step 1]: step timed out (it took 594 ms instead of 580 ms)
10:33:08	[D]	[AirCon:1367]	
0000232934: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
10:33:08	[D]	[AirCon:1651]	
0000232981: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
10:33:11	[D]	[AirCon:1367]	
0000235896: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
10:33:11	[D]	[AirCon:1651]	
0000235942: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
10:33:14	[D]	[AirCon:1651]	
0000238371: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
10:33:14	[D]	[AirCon:1367]	
0000238883: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
10:33:14	[D]	[AirCon:1651]	
0000238930: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
10:33:14	[W]	[AirCon:1067]	
Sequence  [step 1]: step timed out (it took 585 ms instead of 580 ms)
10:33:17	[D]	[AirCon:1367]	
0000241868: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
10:33:17	[D]	[AirCon:1651]	
0000241914: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
10:33:20	[D]	[AirCon:1367]	
0000244845: [<=] [BB 00 01 00 00 00 00 00] [43 FF] 
10:33:20	[D]	[AirCon:1651]	
0000244892: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58] 
10:33:21	[D]	[sensor:094]	
'airco WiFi Signal': Sending state -58.00000 dBa with 0 decimals of accuracy
10:33:21	[D]	[AirCon:1651]	
0000245391: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E] 
10:33:22	[W]	[AirCon:1067]	
Sequence  [step 1]: step timed out (it took 594 ms instead of 580 ms)
10:33:22	[D]	[AirCon:1367]	
0000246820: [<=] [BB 00 01 00 00 00 00 00] [43 FF] `

Can you see why?

PeJeWe, 2024-01-13:
Okay, I will check the wiring

PeJeWe, 2024-01-13:
How come that i don't see the fan speed settings in the esphome web userinterface.
And also the health settings.
I have used the settings from the ac_common file.
The ones I intend to use is.
Outside temp
Inside temp.
AC settings like heat, cool, fan, etc.
Fan speed settings.
Swing modus.
Settings for health, clean, etc
I don't see them all.
It is supporting this, because I can activate it with my IR remote.

Are they switches, or buttons.
Maybe you can help me with this

GrKoR, 2024-01-13:
ESPHome web interface isn't useful. ESPHome is designed to be used with Home Assistant. All features are available in the HA but not in the esphome web interface.
You can implement buttons/switches for manual features activation. But you need go deeper in esphome and C/C++.

PeJeWe, 2024-01-13:
I can confirm that AC1 airconditioning is working with you're software.

I can also use it with domoticz by mqtt.

Can you help me with sending mqtt commands to the esphome controller so I can use those functions?

GrKoR, 2024-01-13:
I can't help you, sorry. I don't use MQTT and I haven't experience with it. Try to find discussions about esphome+mqtt+custom_presets on home Assistant community site.