GrKoR/esphome_aux_ac_component #95: AC Outdoor Temperature wrong?

Steamerzone, 2024-02-26:
It looks like the reported AC Outdoor Temperature is wrong, if I haven't made an error in my setup.

It reports the same value as AC Indoor Coil Temperature.

Emile

GrKoR, 2024-02-27:
Hello!
Other setups don't confirm this. Without logs I can't check your situation. Maybe your AC reports the same value because the AC is off and both sensors has the same temperature.

Steamerzone, 2024-02-27:
Hello!

Both outdoor_ambient_temperature and indoor_coil_temperature report the same temperature of 35'C degree in my setup.
The indoor_coil_temperature seems to be correct, outdoor_ambient_temperature not, it's about 4'C outside. 

Logs after firmware update/reboot:

`[14:12:50][I][app:102]: ESPHome version 2024.2.1 compiled on Feb 27 2024, 14:11:32
[14:12:50][C][wifi:577]: WiFi:
[14:12:50][C][wifi:414]:   SSID: [redacted]
[14:12:50][C][wifi:415]:   IP Address: 10.0.1.231
[14:12:50][C][wifi:416]:   BSSID: [redacted]
[14:12:50][C][wifi:418]:   Hostname: 'esp8266-livingroom-ac'
[14:12:50][C][wifi:420]:   Signal strength: -42 dB ▂▄▆█
[14:12:50][C][wifi:424]:   Channel: 1
[14:12:50][C][wifi:425]:   Subnet: 255.255.255.0
[14:12:50][C][wifi:426]:   Gateway: 10.0.1.100
[14:12:50][C][wifi:427]:   DNS1: 10.0.1.100
[14:12:50][C][wifi:428]:   DNS2: 0.0.0.0
[14:12:50][C][logger:447]: Logger:
[14:12:50][C][logger:448]:   Level: DEBUG
[14:12:50][C][logger:449]:   Log Baud Rate: 0
[14:12:50][C][logger:451]:   Hardware UART: UART0
[14:12:50][C][uart.arduino_esp8266:118]: UART Bus:
[14:12:50][C][uart.arduino_esp8266:119]:   TX Pin: GPIO1
[14:12:50][C][uart.arduino_esp8266:120]:   RX Pin: GPIO3
[14:12:50][C][uart.arduino_esp8266:122]:   RX Buffer Size: 256
[14:12:50][C][uart.arduino_esp8266:124]:   Baud Rate: 4800 baud
[14:12:50][C][uart.arduino_esp8266:125]:   Data Bits: 8
[14:12:50][C][uart.arduino_esp8266:126]:   Parity: EVEN
[14:12:50][C][uart.arduino_esp8266:127]:   Stop bits: 1
[14:12:50][C][uart.arduino_esp8266:129]:   Using hardware serial interface.
[14:12:50][C][template.number:050]: Template Number 'Vertical Louvers'
[14:12:50][C][template.number:050]:   Icon: 'mdi:circle-small'
[14:12:50][C][template.number:051]:   Optimistic: NO
[14:12:50][C][template.number:052]:   Update Interval: 30.0s
[14:12:50][D][AirCon:338]: 0000038997: [<=] [BB 00 07 00 00 00 18 00] 01 21 E0 81 00 04 65 30 43 43 43 64 24 20 50 39 2C 08 A6 F0 13 00 00 01 [FE 2C]
[14:12:50][D][climate:396]: 'Livingroom AC' - Sending state:
[14:12:50][D][climate:399]:   Mode: HEAT
[14:12:50][D][climate:404]:   Fan Mode: MEDIUM
[14:12:50][D][climate:416]:   Swing Mode: OFF
[14:12:50][D][climate:419]:   Current Temperature: 16.10°C
[14:12:50][D][climate:425]:   Target Temperature: 16.00°C
[14:12:50][D][sensor:093]: 'AC Inverter Power': Sending state 44.00000 % with 0 decimals of accuracy
[14:12:50][C][template.number:050]: Template Number 'IPower Limit Value'
[14:12:50][C][template.number:050]:   Icon: 'mdi:battery-unknown'
[14:12:50][C][template.number:051]:   Optimistic: NO
[14:12:50][C][template.number:052]:   Update Interval: 60.0s
[14:12:50][C][template.switch:068]: Template Switch 'Display'
[14:12:50][C][template.switch:090]:   Restore Mode: always OFF
[14:12:50][C][template.switch:057]:   Optimistic: NO
[14:12:50][C][AirCon:357]: AUX HVAC:
[14:12:50][C][AirCon:358]: firmware version: 1.0.0
[14:12:50][C][AirCon:571]: ClimateTraits:
[14:12:50][C][AirCon:572]:   [x] Visual settings:
[14:12:50][C][AirCon:573]:       - Min temperature: 16.0
[14:12:50][C][AirCon:574]:       - Max temperature: 32.0
[14:12:50][C][AirCon:575]:       - Temperature step:
[14:12:50][C][AirCon:576]:           Target: 1.0
[14:12:50][C][AirCon:577]:           Current: 1.0
[14:12:50][C][AirCon:578]:       - Min humidity: 30
[14:12:50][C][AirCon:579]:       - Max humidity: 99
[14:12:50][C][AirCon:581]:   [x] Supports current temperature
[14:12:50][C][AirCon:596]:   [x] Supported modes:
[14:12:50][C][AirCon:598]:       - OFF
[14:12:50][C][AirCon:598]:       - HEAT_COOL
[14:12:50][C][AirCon:598]:       - COOL
[14:12:50][C][AirCon:598]:       - HEAT
[14:12:50][C][AirCon:598]:       - FAN_ONLY
[14:12:50][C][AirCon:598]:       - DRY
[... 251 lines cut ...]
[14:14:21][D][climate:399]:   Mode: HEAT
[14:14:21][D][climate:404]:   Fan Mode: MEDIUM
[14:14:21][D][climate:416]:   Swing Mode: OFF
[14:14:21][D][climate:419]:   Current Temperature: 16.10°C
[14:14:21][D][climate:425]:   Target Temperature: 16.00°C
[14:14:21][D][sensor:093]: 'AC Indoor Ambient Temperature': Sending state 16.10000 °C with 1 decimals of accuracy
[14:14:24][D][AirCon:338]: 0000133036: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[14:14:24][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[14:14:25][D][number:012]: 'Vertical Louvers': Sending state 2.000000
[14:14:27][D][AirCon:338]: 0000136013: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[14:14:27][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[14:14:28][D][AirCon:124]: 0000017283: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E]
[14:14:28][D][AirCon:338]: 0000136881: [<=] [BB 00 07 00 00 00 0F 00] 01 11 41 34 14 40 00 80 00 00 20 00 10 00 00 [A7 F9]
[14:14:28][D][AirCon:124]: 0000017284: [=>] [BB 00 06 80 00 00 02 00] 21 01 [1B 7E]
[14:14:28][D][AirCon:338]: 0000137123: [<=] [BB 00 07 00 00 00 18 00] 01 21 E0 81 00 04 65 30 43 43 43 64 24 20 50 39 2C 08 A4 F0 13 00 00 01 [00 2D]
[14:14:31][D][AirCon:338]: 0000140039: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[14:14:31][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[14:14:34][D][AirCon:338]: 0000143024: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[14:14:34][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[14:14:35][D][AirCon:124]: 0000017283: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E]
[14:14:35][D][AirCon:338]: 0000143875: [<=] [BB 00 07 00 00 00 0F 00] 01 11 41 34 14 40 00 80 00 00 20 00 10 00 00 [A7 F9]
[14:14:35][D][AirCon:124]: 0000017284: [=>] [BB 00 06 80 00 00 02 00] 21 01 [1B 7E]
[14:14:35][D][AirCon:338]: 0000144118: [<=] [BB 00 07 00 00 00 18 00] 01 21 E0 81 00 04 65 30 43 43 43 64 24 20 50 39 2C 08 A4 F0 11 00 00 01 [02 2D]
[14:14:38][D][AirCon:338]: 0000147033: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[14:14:38][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]`

drupry, 2024-03-04:
I can confirm this, ** AC Outdoor Ambient Temperature** is reporting the same temperature as ** AC Indoor Coil Temperature** witch seems OK. Its about 10°C outdoor here.
[image]

closed 2024-03-04 by GrKoR