GrKoR/esphome_aux_ac_component #96: Disable fan mute in the same fan mode doesn't seem to work.

Steamerzone, 2024-02-27:
Hello,

Disabling 'Custom Fan Mode: MUTE' with no change in fan mode doesn't seem to work,

fan auto -> fan mute -> fan medium   works
fan auto -> fan mute -> fan auto        doesn't work

Emile

GrKoR, 2024-02-27:
Hi, Emile!

Please could you send me a esphome log, where you are changing the fan mode?

Steamerzone, 2024-02-27:
Hello,

Of course here's the log, at the end it's on mode medium/mute and I want to switch to medium/no mute.

[18:55:15][D][AirCon:338]: 0016983831: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[18:55:15][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[18:55:16][D][AirCon:124]: 0000017283: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E]
[18:55:16][D][AirCon:338]: 0016984715: [<=] [BB 00 07 00 00 00 0F 00] 01 11 6B 21 03 A0 00 80 00 00 20 00 10 64 00 [8E 48]
[18:55:16][D][AirCon:124]: 0000017284: [=>] [BB 00 06 80 00 00 02 00] 21 01 [1B 7E]
[18:55:16][D][AirCon:338]: 0016984957: [<=] [BB 00 07 00 00 00 18 00] 01 21 E0 81 00 06 76 31 4F 4F 4F 64 23 1F 6E 39 58 18 AE 40 2F 00 00 05 [68 BA]
[18:55:16][D][climate:396]: 'Livingroom AC' - Sending state:
[18:55:16][D][climate:399]:   Mode: HEAT
[18:55:16][D][climate:404]:   Fan Mode: AUTO
[18:55:16][D][climate:407]:   Custom Fan Mode: 
[18:55:16][D][climate:416]:   Swing Mode: OFF
[18:55:16][D][climate:419]:   Current Temperature: 17.50°C
[18:55:16][D][climate:425]:   Target Temperature: 21.00°C
[18:55:16][D][sensor:093]: 'AC Inverter Power': Sending state 88.00000 % with 0 decimals of accuracy
[18:55:19][D][AirCon:338]: 0016987872: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[18:55:19][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[18:55:19][D][climate:011]: 'Livingroom AC' - Setting
[18:55:19][D][climate:019]:  Custom Fan: MUTE
[18:55:19][D][climate:396]: 'Livingroom AC' - Sending state:
[18:55:19][D][climate:399]:   Mode: HEAT
[18:55:19][D][climate:404]:   Fan Mode: AUTO
[18:55:19][D][climate:407]:   Custom Fan Mode: MUTE
[18:55:19][D][climate:416]:   Swing Mode: OFF
[18:55:19][D][climate:419]:   Current Temperature: 17.50°C
[18:55:19][D][climate:425]:   Target Temperature: 21.00°C
[18:55:19][D][AirCon:124]: 0016987949: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E]
[18:55:19][D][AirCon:338]: 0016988170: [<=] [BB 00 07 00 00 00 0F 00] 01 11 6B 21 03 A0 00 80 00 00 20 00 10 64 00 [8E 48]
[18:55:19][D][climate:396]: 'Livingroom AC' - Sending state:
[18:55:19][D][climate:399]:   Mode: HEAT
[18:55:19][D][climate:404]:   Fan Mode: AUTO
[18:55:19][D][climate:407]:   Custom Fan Mode: 
[18:55:19][D][climate:416]:   Swing Mode: OFF
[18:55:19][D][climate:419]:   Current Temperature: 17.50°C
[18:55:19][D][climate:425]:   Target Temperature: 21.00°C
[18:55:19][D][AirCon:124]: 0016988203: [=>] [BB 00 06 80 00 00 0F 00] 01 01 6B 21 03 A0 80 80 00 00 20 00 10 64 00 [0E D8]
[18:55:20][D][AirCon:338]: 0016988423: [<=] [BB 00 07 00 00 00 04 00] 01 01 0E D8 [2A 26]
[18:55:20][D][AirCon:124]: 0000017283: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E]
[18:55:20][D][AirCon:338]: 0016988664: [<=] [BB 00 07 00 00 00 0F 00] 01 11 6B 21 03 A0 80 80 00 00 20 00 10 64 00 [0E 48]
[18:55:20][D][climate:396]: 'Livingroom AC' - Sending state:
[18:55:20][D][climate:399]:   Mode: HEAT
[18:55:20][D][climate:404]:   Fan Mode: AUTO
[18:55:20][D][climate:407]:   Custom Fan Mode: MUTE
[18:55:20][D][climate:416]:   Swing Mode: OFF
[18:55:20][D][climate:419]:   Current Temperature: 17.50°C
[18:55:20][D][climate:425]:   Target Temperature: 21.00°C
[18:55:20][D][AirCon:124]: 0000017284: [=>] [BB 00 06 80 00 00 02 00] 21 01 [1B 7E]
[18:55:20][D][AirCon:338]: 0016988915: [<=] [BB 00 07 00 00 00 18 00] 01 21 E0 81 00 02 75 31 4F 4F 4F 64 23 1F 6F 39 58 18 AE 40 2D 00 00 05 [6A BE]
[18:55:20][D][climate:396]: 'Livingroom AC' - Sending state:
[18:55:20][D][climate:399]:   Mode: HEAT
[18:55:20][D][climate:404]:   Fan Mode: AUTO
[18:55:20][D][climate:407]:   Custom Fan Mode: MUTE
[18:55:20][D][climate:416]:   Swing Mode: OFF
[18:55:20][D][climate:419]:   Current Temperature: 17.50°C
[18:55:20][D][climate:425]:   Target Temperature: 21.00°C
[18:55:20][D][sensor:093]: 'AC Compressor Discharge Temperature': Sending state 79.00000 °C with 0 decimals of accuracy
[18:55:23][D][AirCon:124]: 0000017283: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E]
[18:55:23][D][AirCon:338]: 0016991722: [<=] [BB 00 07 00 00 00 0F 00] 01 11 6B 21 03 A0 80 80 00 00 20 00 10 64 00 [0E 48]
[18:55:23][D][AirCon:124]: 0000017284: [=>] [BB 00 06 80 00 00 02 00] 21 01 [1B 7E]
[18:55:23][D][AirCon:338]: 0016991964: [<=] [BB 00 07 00 00 00 18 00] 01 21 E0 81 00 02 50 31 4F 4F 4F 64 23 1F 6F 39 58 18 AE 40 2B 00 00 05 [91 BE]
[18:55:24][D][climate:011]: 'Livingroom AC' - Setting
[18:55:24][D][climate:024]:   Fan: MEDIUM
[18:55:24][D][climate:396]: 'Livingroom AC' - Sending state:
[18:55:24][D][climate:399]:   Mode: HEAT
[18:55:24][D][climate:404]:   Fan Mode: MEDIUM
[18:55:24][D][climate:407]:   Custom Fan Mode: MUTE
[18:55:24][D][climate:416]:   Swing Mode: OFF
[... 90 lines cut ...]
[18:55:35][D][climate:024]:   Fan: MEDIUM
[18:55:36][D][AirCon:338]: 0017004844: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[18:55:36][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[18:55:37][D][AirCon:124]: 0000017283: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E]
[18:55:37][D][AirCon:338]: 0017005729: [<=] [BB 00 07 00 00 00 0F 00] 01 11 6B 21 03 40 80 80 00 00 20 00 10 64 00 [0E A8]
[18:55:37][D][AirCon:124]: 0000017284: [=>] [BB 00 06 80 00 00 02 00] 21 01 [1B 7E]
[18:55:37][D][AirCon:338]: 0017005971: [<=] [BB 00 07 00 00 00 18 00] 01 21 E0 81 00 02 56 31 50 50 50 64 23 1E 70 39 4E 16 AE 40 28 00 00 06 [95 BF]
[18:55:37][D][climate:396]: 'Livingroom AC' - Sending state:
[18:55:37][D][climate:399]:   Mode: HEAT
[... cut ...]
[18:55:37][D][sensor:093]: 'AC Inverter Power': Sending state 78.00000 % with 0 decimals of accuracy
[18:55:38][D][climate:011]: 'Livingroom AC' - Setting
[18:55:38][D][climate:024]:   Fan: MEDIUM
[18:55:40][D][AirCon:338]: 0017008882: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[18:55:40][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[18:55:43][D][AirCon:338]: 0017011869: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[18:55:43][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]

GrKoR, 2024-02-27:
Thank you. Will check it soon

drupry, 2024-03-04:
I can confirm this. After update, selecting auto after mute is not working. 

https://github.com/GrKoR/esphome_aux_ac_component/assets/124905549/5f4fb8d2-fe98-402c-a675-cf089e946fd4

GrKoR, 2024-03-04:
@Steamerzone and @drupry could you play with 'TURBO' fan mode? Is it has the same bug?

Please, send me logs for `fan auto -> fan mute -> fan auto` and `fan auto -> fan turbo -> fan auto` sequences.

Steamerzone, 2024-03-04:
That does seem to have the same bug. 

[21:57:58][D][AirCon:338]: 0002518762: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[21:57:58][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[21:57:59][D][AirCon:124]: 0000017352: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E]
[21:57:59][D][AirCon:338]: 0002519647: [<=] [BB 00 07 00 00 00 0F 00] 01 11 63 2B 1A 40 00 80 00 00 20 00 10 00 00 [80 02]
[21:57:59][D][AirCon:124]: 0000017353: [=>] [BB 00 06 80 00 00 02 00] 21 01 [1B 7E]
[21:57:59][D][AirCon:338]: 0002519888: [<=] [BB 00 07 00 00 00 18 00] 01 21 E0 81 00 04 65 33 4A 4A 4A 64 28 23 5D 39 31 0A B0 F0 14 00 00 09 [CF 15]
[21:58:02][D][AirCon:338]: 0002522805: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[21:58:02][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[21:58:05][D][AirCon:338]: 0002525791: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[21:58:05][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[21:58:06][D][AirCon:124]: 0000017352: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E]
[21:58:06][D][AirCon:338]: 0002526659: [<=] [BB 00 07 00 00 00 0F 00] 01 11 63 2B 1A 40 00 80 00 00 20 00 10 00 00 [80 02]
[21:58:06][D][AirCon:124]: 0000017353: [=>] [BB 00 06 80 00 00 02 00] 21 01 [1B 7E]
[21:58:06][D][AirCon:338]: 0002526901: [<=] [BB 00 07 00 00 00 18 00] 01 21 E0 81 00 04 65 33 4A 4A 4A 64 28 23 5D 39 31 0A AE F0 19 00 00 09 [CC 15]
[21:58:08][D][climate:011]: 'Livingroom AC' - Setting
[21:58:08][D][climate:024]:   Fan: AUTO
[21:58:08][D][climate:396]: 'Livingroom AC' - Sending state:
[21:58:08][D][climate:399]:   Mode: HEAT
[21:58:08][D][climate:404]:   Fan Mode: AUTO
[21:58:08][D][climate:407]:   Custom Fan Mode: 
[21:58:08][D][climate:416]:   Swing Mode: OFF
[21:58:08][D][climate:419]:   Current Temperature: 19.90°C
[21:58:08][D][climate:425]:   Target Temperature: 20.00°C
[21:58:08][D][AirCon:124]: 0002528466: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E]
[21:58:08][D][AirCon:338]: 0002528686: [<=] [BB 00 07 00 00 00 0F 00] 01 11 63 2B 1A 40 00 80 00 00 20 00 10 00 00 [80 02]
[21:58:08][D][climate:396]: 'Livingroom AC' - Sending state:
[21:58:08][D][climate:399]:   Mode: HEAT
[21:58:08][D][climate:404]:   Fan Mode: MEDIUM
[21:58:08][D][climate:407]:   Custom Fan Mode: 
[21:58:08][D][climate:416]:   Swing Mode: OFF
[21:58:08][D][climate:419]:   Current Temperature: 19.90°C
[21:58:08][D][climate:425]:   Target Temperature: 20.00°C
[21:58:08][D][AirCon:124]: 0002528719: [=>] [BB 00 06 80 00 00 0F 00] 01 01 63 2B 1A A0 00 80 00 00 20 00 10 00 00 [80 32]
[21:58:09][D][AirCon:338]: 0002528939: [<=] [BB 00 07 00 00 00 04 00] 01 01 80 32 [B8 CB]
[21:58:09][D][AirCon:124]: 0000017352: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E]
[21:58:09][D][AirCon:338]: 0002529181: [<=] [BB 00 07 00 00 00 0F 00] 01 11 63 2B 1A A0 00 80 00 00 20 00 10 00 00 [7F A2]
[21:58:09][D][climate:396]: 'Livingroom AC' - Sending state:
[21:58:09][D][climate:399]:   Mode: HEAT
[21:58:09][D][climate:404]:   Fan Mode: AUTO
[21:58:09][D][climate:407]:   Custom Fan Mode: 
[21:58:09][D][climate:416]:   Swing Mode: OFF
[21:58:09][D][climate:419]:   Current Temperature: 19.90°C
[21:58:09][D][climate:425]:   Target Temperature: 20.00°C
[21:58:09][D][AirCon:124]: 0000017353: [=>] [BB 00 06 80 00 00 02 00] 21 01 [1B 7E]
[21:58:09][D][AirCon:338]: 0002529434: [<=] [BB 00 07 00 00 00 18 00] 01 21 E0 81 00 06 65 33 4A 4A 4A 64 28 23 5D 39 31 0B AE F0 17 00 00 09 [CE 12]
[21:58:09][D][climate:396]: 'Livingroom AC' - Sending state:
[21:58:09][D][climate:399]:   Mode: HEAT
[21:58:09][D][climate:404]:   Fan Mode: AUTO
[21:58:09][D][climate:407]:   Custom Fan Mode: 
[21:58:09][D][climate:416]:   Swing Mode: OFF
[21:58:09][D][climate:419]:   Current Temperature: 19.90°C
[21:58:09][D][climate:425]:   Target Temperature: 20.00°C
[21:58:10][D][climate:011]: 'Livingroom AC' - Setting
[21:58:10][D][climate:019]:  Custom Fan: MUTE
[21:58:10][D][climate:396]: 'Livingroom AC' - Sending state:
[21:58:10][D][climate:399]:   Mode: HEAT
[21:58:10][D][climate:404]:   Fan Mode: AUTO
[21:58:10][D][climate:407]:   Custom Fan Mode: MUTE
[21:58:10][D][climate:416]:   Swing Mode: OFF
[21:58:10][D][climate:419]:   Current Temperature: 19.90°C
[21:58:10][D][climate:425]:   Target Temperature: 20.00°C
[21:58:10][D][AirCon:124]: 0002530252: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E]
[21:58:10][D][AirCon:338]: 0002530472: [<=] [BB 00 07 00 00 00 0F 00] 01 11 63 2B 1A A0 00 80 00 00 20 00 10 00 00 [7F A2]
[21:58:10][D][climate:396]: 'Livingroom AC' - Sending state:
[21:58:10][D][climate:399]:   Mode: HEAT
[21:58:10][D][climate:404]:   Fan Mode: AUTO
[21:58:10][D][climate:407]:   Custom Fan Mode: 
[21:58:10][D][climate:416]:   Swing Mode: OFF
[... 263 lines cut ...]
[21:58:37][D][climate:396]: 'Livingroom AC' - Sending state:
[21:58:37][D][climate:399]:   Mode: HEAT
[21:58:37][D][climate:404]:   Fan Mode: MEDIUM
[21:58:37][D][climate:407]:   Custom Fan Mode: 
[21:58:37][D][climate:416]:   Swing Mode: OFF
[21:58:37][D][climate:419]:   Current Temperature: 19.90°C
[21:58:37][D][climate:425]:   Target Temperature: 20.00°C
[21:58:37][D][sensor:093]: 'AC Compressor Suction Temperature': Sending state 3.00000 °C with 0 decimals of accuracy
[21:58:37][D][sensor:093]: 'AC Inverter Power': Sending state 50.00000 % with 0 decimals of accuracy
[21:58:40][D][AirCon:338]: 0002560240: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[... cut ...]
[21:58:42][D][climate:407]:   Custom Fan Mode: 
[21:58:42][D][climate:416]:   Swing Mode: OFF
[21:58:42][D][climate:419]:   Current Temperature: 19.90°C
[21:58:42][D][climate:425]:   Target Temperature: 20.00°C
[21:58:42][D][sensor:093]: 'AC Indoor Coil Temperature': Sending state 42.00000 °C with 0 decimals of accuracy
[21:58:42][D][sensor:093]: 'AC Outdoor Ambient Temperature': Sending state 42.00000 °C with 0 decimals of accuracy
[21:58:42][D][sensor:093]: 'AC Inverter Power': Sending state 49.00000 % with 0 decimals of accuracy

closed 2024-11-03 by GrKoR