GrKoR/esphome_aux_ac_component #97: All temperature sensors are NA for Ballu AC (ESP32)

GrKoR, 2024-03-04:
[image]

Firmware version: v.1.0.0
Details: ask `Сергей Владимиров` in telegram

### Part of YAML config:
```yaml
esp32:
  board: esp32dev
  framework:
#    type: arduino
    type: esp-idf

climate:
  - platform: aux_ac
    name: ${upper_devicename}
    id: aux_id
    uart_id: ac_uart_bus
    period: 7s
    display_inverted: true
    optimistic: true
    display_state:
      name: ${upper_devicename} Display State
      id: ${devicename}_display_state
      internal: false
    indoor_ambient_temperature:
      name: ${upper_devicename} Indoor Ambient Temperature
      id: ${devicename}_indoor_ambient_temp
      internal: false
    indoor_coil_temperature:
      name: ${upper_devicename} Indoor Coil Temperature
      id: ${devicename}_indoor_coil_temp
      internal: false
    outdoor_ambient_temperature:
      name: ${upper_devicename} Outdoor Ambient Temperature
      id: ${devicename}_outdoor_ambient_temp
      internal: false
    outdoor_condenser_temperature:
      name: ${upper_devicename} Outdoor Condenser Temperature
      id: ${devicename}_outdoor_condenser_temp
      internal: false
    compressor_suction_temperature:
      name: ${upper_devicename} Compressor Suction Temperature
      id: ${devicename}_compressor_suction_temp
      internal: false
    compressor_discharge_temperature:
      name: ${upper_devicename} Compressor Discharge Temperature
      id: ${devicename}_compressor_discharge_temp
      internal: false
    defrost_temperature:
      name: ${upper_devicename} Defrost Temperature
      id: ${devicename}_defrost_temp
      internal: false
    defrost_state:
      name: ${upper_devicename} Defrost State
      id: ${devicename}_defrost_state
      internal: false
    inverter_power:
      name: ${upper_devicename} Inverter Power
      id: ${devicename}_inverter_power
      internal: false
    preset_reporter:
      name: ${upper_devicename} Preset Reporter
      id: ${devicename}_preset_reporter
      internal: false
    vlouver_state:
      name: ${upper_devicename} VLouvers State
      id: ${devicename}_vlouver_state
      internal: false
    visual:
      min_temperature: 16
      max_temperature: 32
      temperature_step: 1
#        target_temperature: 0.5
#        current_temperature: 0.1
    supported_modes:
      - HEAT_COOL
      - COOL
      - HEAT
      - DRY
      - FAN_ONLY
    custom_fan_modes:
      - MUTE
      - TURBO
    supported_presets:
      - SLEEP
    custom_presets:
      - CLEAN
      - HEALTH
      - ANTIFUNGUS
    supported_swing_modes:
      - VERTICAL
      - HORIZONTAL
      - BOTH
```

GrKoR, 2024-03-04:
### Logs:
```
[11:01:30][C][AirCon:357]: AUX HVAC:
[11:01:30][C][AirCon:358]: firmware version: 1.0.0
[11:01:30][C][AirCon:571]: ClimateTraits:
[11:01:30][C][AirCon:572]:   [x] Visual settings:
[11:01:30][C][AirCon:573]:       - Min temperature: 16.0
[11:01:30][C][AirCon:574]:       - Max temperature: 32.0
[11:01:30][C][AirCon:575]:       - Temperature step:
[11:01:30][C][AirCon:576]:           Target: 1.0
[11:01:30][C][AirCon:577]:           Current: 1.0
[11:01:30][C][AirCon:578]:       - Min humidity: 30
[11:01:30][C][AirCon:579]:       - Max humidity: 99
[11:01:30][C][AirCon:581]:   [x] Supports current temperature
[11:01:30][C][AirCon:596]:   [x] Supported modes:
[11:01:30][C][AirCon:598]:       - OFF
[11:01:30][C][AirCon:598]:       - HEAT_COOL
[11:01:30][C][AirCon:598]:       - COOL
[11:01:30][C][AirCon:598]:       - HEAT
[11:01:30][C][AirCon:364]:   Preset Reporter 'AC Ballu Preset Reporter'
[11:01:30][C][AirCon:364]:     Icon: 'mdi:format-list-group'
[11:01:30][C][AirCon:368]:       Device Class: 'temperature'
[11:01:30][C][AirCon:368]:       State Class: 'measurement'
[11:01:30][C][AirCon:369]:       Accuracy Decimals: 0
[11:01:30][C][AirCon:369]:       Icon: 'mdi:home-thermometer-outline'
[11:01:30][D][number:012]: 'AC Ballu Vertical Louvers': Sending state 6.000000
[11:01:30][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:01:30][V][json:058]: Size after shrink 108 bytes
[11:01:30][D][AirCon:338]: 0000005701: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[11:01:30][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[11:01:32][D][AirCon:124]: 0000000046: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E]
[11:01:32][D][AirCon:338]: 0000007205: [<=] [BB 00 07 00 00 00 0F 00] 01 11 6D E4 38 20 00 80 00 00 20 00 00 00 00 [67 69]
[11:01:32][D][AirCon:124]: 0000000046: [=>] [BB 00 06 80 00 00 02 00] 21 01 [1B 7E]
[11:01:32][D][AirCon:338]: 0000007434: [<=] [BB 00 07 00 00 00 19 00] 01 21 E0 81 00 02 5A 34 3B 3B 3B 64 0E 11 69 39 33 0B C0 28 14 00 00 04 00 [F4 03]
[11:01:32][D][number:012]: 'AC Ballu Vertical Louvers': Sending state 6.000000
[11:01:32][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:01:32][V][json:058]: Size after shrink 108 bytes
[11:01:34][D][AirCon:338]: 0000008909: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[11:01:34][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[11:01:34][D][number:012]: 'AC Ballu Vertical Louvers': Sending state 6.000000
[11:01:34][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:01:34][V][json:058]: Size after shrink 108 bytes
[11:01:36][D][number:012]: 'AC Ballu Vertical Louvers': Sending state 6.000000
[11:01:36][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:01:36][V][json:058]: Size after shrink 108 bytes
[11:01:37][D][AirCon:338]: 0000011920: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[11:01:37][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[11:01:38][D][number:012]: 'AC Ballu Vertical Louvers': Sending state 6.000000
[11:01:38][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:01:38][V][json:058]: Size after shrink 108 bytes
[11:01:39][D][AirCon:124]: 0000000046: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E]
[11:01:39][D][AirCon:338]: 0000014230: [<=] [BB 00 07 00 00 00 0F 00] 01 11 6D E4 38 20 00 80 00 00 20 00 00 00 00 [67 69]
[11:01:39][D][AirCon:124]: 0000000046: [=>] [BB 00 06 80 00 00 02 00] 21 01 [1B 7E]
[11:01:39][D][AirCon:338]: 0000014497: [<=] [BB 00 07 00 00 00 19 00] 01 21 E0 81 00 02 5A 34 3C 3C 3C 64 0E 10 69 39 33 0B C0 28 14 00 00 03 00 [F2 04]
[11:01:40][D][AirCon:338]: 0000015014: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[11:01:40][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[11:01:40][D][number:012]: 'AC Ballu Vertical Louvers': Sending state 6.000000
[11:01:40][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:01:40][V][json:058]: Size after shrink 108 bytes
[11:01:42][D][number:012]: 'AC Ballu Vertical Louvers': Sending state 6.000000
[11:01:42][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:01:42][V][json:058]: Size after shrink 108 bytes
[11:01:43][D][AirCon:338]: 0000018000: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[11:01:43][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[11:01:44][D][number:012]: 'AC Ballu Vertical Louvers': Sending state 6.000000
[11:01:44][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:01:44][V][json:058]: Size after shrink 108 bytes
[11:01:46][D][AirCon:338]: 0000021004: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[11:01:46][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[11:01:46][D][AirCon:124]: 0000000046: [=>] [BB 00 06 80 00 00 02 00] 11 01 [2B 7E]
[... 1244 lines cut ...]
[11:08:22][D][number:012]: 'AC Ballu Vertical Louvers': Sending state 6.000000
[11:08:22][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:08:22][V][json:058]: Size after shrink 108 bytes
[11:08:23][D][AirCon:338]: 0000417856: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[11:08:23][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[11:08:24][D][number:012]: 'AC Ballu Vertical Louvers': Sending state 6.000000
[... cut ...]
[11:08:28][D][number:012]: 'AC Ballu Vertical Louvers': Sending state 6.000000
[11:08:28][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
[11:08:28][V][json:058]: Size after shrink 108 bytes
[11:08:29][D][AirCon:338]: 0000423952: [<=] [BB 00 01 00 00 00 00 00] [43 FF]
[11:08:29][D][AirCon:124]: 0000000000: [=>] [BB 00 01 80 01 00 08 00] 1C 27 00 00 00 00 00 00 [1E 58]
[11:08:30][D][number:012]: 'AC Ballu Vertical Louvers': Sending state 6.000000
[11:08:30][V][json:038]: Attempting to allocate 512 bytes for JSON serialization
```

gijsje, 2024-03-29:
I have the same issue that most sensors are not show after an update to ESP32

GrKoR, 2024-04-04:
@gijsje hi!
This issue is ballu air conditioners specific. Most likely you have another issue.
Component v.1.0.0 has several bugs which I can't fix in a short time. Try install previous version of the component ([link](https://github.com/GrKoR/esphome_aux_ac_component/issues/101#issuecomment-1980226807)

tomadam123, 2024-05-16:
Test the temperature sensors individually to ensure they are functioning correctly. You can do this by using a multimeter to measure the resistance or voltage output of each sensor. Check the controller or thermostat unit for any signs of damage or malfunction. If possible, try using a different controller to see if the issue persists. [click to investigate](https://airconditioningservicesnj.com/)

GrKoR, 2024-05-17:
@tomadam123 Hey!
This issue is about unusual frame size for some AUX models (rare situation). And it's related v.1.0.0 The f the component.
Your comment doesn't match the topic, am I right?

GrKoR, 2024-05-21:
The @tomadam123 comment was hidden.
Reasons:
* it looks like spam (strange link in the message);
* author doesn't reply for my question but posted another one comment with similar content in the other thread.

closed 2024-11-03 by GrKoR