GrKoR/esphome_aux_ac_component #98: Display State, Defrost State and some other sensors are 'NA' (v.1.0.0)

GrKoR, 2024-03-04:
firmware version: v.1.0.0

[image] [image]

Logs: see messages from Ahmad Al Salhi  in telegram

GrKoR, 2024-03-04:
Author: `Андрей Александрович`
> минутку. сравнил свою старую конфигурации и новую. у меня в старой `display_inverted: false` а в новой `display_inverted: true`
> что-то изменилось в этом плане?
> то есть установить как и было в моём старом конфиге?
> поставил как у в старом конфиге. пробую.
> это победили
> показывает теперь что выключен дисплей
> AC Defrost State это узнаю когда запущу отопление.

`Display State` was configured incorrect by `Андрей Александрович`. Now it works.

drupry, 2024-03-08:
i have the same bug even though I have in my config `display_inverted: false` .
 `defrost state` and `AC Inverter Power Limit State` is unknown. Sometimes also `AC Compressor Suction Temperature` is unknown because the temperature goes below zero but i noticed its already open in another thread. 
[image]

closed 2024-11-03 by GrKoR