GrKoR/esphome_aux_ac_component #100: Warning in esphome compiler

akdri, 2024-03-05:
alert in the esphome compiler

it's wrong?

Compiling .pioenvs/aire_dormitorio/src/esphome/components/aux_ac/command_builder.cpp.o
In file included from src/esphome.h:17,
                 from src/esphome/components/aux_ac/aircon.h:6,
                 from src/esphome/components/aux_ac/aircon.cpp:1:
src/esphome/components/aux_ac/automation.h: In member function 'void esphome::aux_airconditioner::AirConDisplayOffAction<Ts>::play(Ts ...)':
src/esphome/components/aux_ac/automation.h:20:52: warning: invalid use of incomplete type 'class esphome::aux_airconditioner::AirCon'
             void play(Ts... x) override { this->ac_->action_display_off(); }
                                                    ^~
src/esphome/components/aux_ac/automation.h:11:15: note: forward declaration of 'class esphome::aux_airconditioner::AirCon'
         class AirCon;
               ^~~~~~
src/esphome/components/aux_ac/automation.h: In member function 'void esphome::aux_airconditioner::AirConDisplayOnAction<Ts>::play(Ts ...)':
src/esphome/components/aux_ac/automation.h:32:52: warning: invalid use of incomplete type 'class esphome::aux_airconditioner::AirCon'
             void play(Ts... x) override { this->ac_->action_display_on(); }
                                                    ^~
src/esphome/components/aux_ac/automation.h:11:15: note: forward declaration of 'class esphome::aux_airconditioner::AirCon'
         class AirCon;
               ^~~~~~
src/esphome/components/aux_ac/automation.h: In member function 'void esphome::aux_airconditioner::AirConVLouverSwingAction<Ts>::play(Ts ...)':
src/esphome/components/aux_ac/automation.h:45:52: warning: invalid use of incomplete type 'class esphome::aux_airconditioner::AirCon'
             void play(Ts... x) override { this->ac_->action_set_vlouver_swing(); }
                                                    ^~
src/esphome/components/aux_ac/automation.h:11:15: note: forward declaration of 'class esphome::aux_airconditioner::AirCon'
         class AirCon;
               ^~~~~~
src/esphome/components/aux_ac/automation.h: In member function 'void esphome::aux_airconditioner::AirConVLouverStopAction<Ts>::play(Ts ...)':
src/esphome/components/aux_ac/automation.h:57:52: warning: invalid use of incomplete type 'class esphome::aux_airconditioner::AirCon'
             void play(Ts... x) override { this->ac_->action_set_vlouver_stop(); }
                                                    ^~
src/esphome/components/aux_ac/automation.h:11:15: note: forward declaration of 'class esphome::aux_airconditioner::AirCon'
         class AirCon;
               ^~~~~~
src/esphome/components/aux_ac/automation.h: In member function 'void esphome::aux_airconditioner::AirConVLouverTopAction<Ts>::play(Ts ...)':
src/esphome/components/aux_ac/automation.h:69:52: warning: invalid use of incomplete type 'class esphome::aux_airconditioner::AirCon'
             void play(Ts... x) override { this->ac_->action_set_vlouver_top_position(); }
                                                    ^~
src/esphome/components/aux_ac/automation.h:11:15: note: forward declaration of 'class esphome::aux_airconditioner::AirCon'
         class AirCon;
               ^~~~~~
src/esphome/components/aux_ac/automation.h: In member function 'void esphome::aux_airconditioner::AirConVLouverMiddleAboveAction<Ts>::play(Ts ...)':
src/esphome/components/aux_ac/automation.h:81:52: warning: invalid use of incomplete type 'class esphome::aux_airconditioner::AirCon'
             void play(Ts... x) override { this->ac_->action_set_vlouver_middle_above_position(); }
                                                    ^~
src/esphome/components/aux_ac/automation.h:11:15: note: forward declaration of 'class esphome::aux_airconditioner::AirCon'
         class AirCon;
               ^~~~~~
src/esphome/components/aux_ac/automation.h: In member function 'void esphome::aux_airconditioner::AirConVLouverMiddleAction<Ts>::play(Ts ...)':
src/esphome/components/aux_ac/automation.h:93:52: warning: invalid use of incomplete type 'class esphome::aux_airconditioner::AirCon'
             void play(Ts... x) override { this->ac_->action_set_vlouver_middle_position(); }
                                                    ^~
src/esphome/components/aux_ac/automation.h:11:15: note: forward declaration of 'class esphome::aux_airconditioner::AirCon'
         class AirCon;
               ^~~~~~
src/esphome/components/aux_ac/automation.h: In member function 'void esphome::aux_airconditioner::AirConVLouverMiddleBelowAction<Ts>::play(Ts ...)':
src/esphome/components/aux_ac/automation.h:105:52: warning: invalid use of incomplete type 'class esphome::aux_airconditioner::AirCon'
             void play(Ts... x) override { this->ac_->action_set_vlouver_middle_below_position(); }
                                                    ^~
src/esphome/components/aux_ac/automation.h:11:15: note: forward declaration of 'class esphome::aux_airconditioner::AirCon'
         class AirCon;
               ^~~~~~
src/esphome/components/aux_ac/automation.h: In member function 'void esphome::aux_airconditioner::AirConVLouverBottomAction<Ts>::play(Ts ...)':
src/esphome/components/aux_ac/automation.h:117:52: warning: invalid use of incomplete type 'class esphome::aux_airconditioner::AirCon'
             void play(Ts... x) override { this->ac_->action_set_vlouver_bottom(); }
                                                    ^~
src/esphome/components/aux_ac/automation.h:11:15: note: forward declaration of 'class esphome::aux_airconditioner::AirCon'
         class AirCon;
               ^~~~~~
src/esphome/components/aux_ac/automation.h: In member function 'void esphome::aux_airconditioner::AirConVLouverSetAction<Ts>::play(Ts ...)':
src/esphome/components/aux_ac/automation.h:132:26: warning: invalid use of incomplete type 'class esphome::aux_airconditioner::AirCon'
                 this->ac_->action_set_vlouver_position((vlouver_esphome_position_t)this->value_.value(x...));
                          ^~
src/esphome/components/aux_ac/automation.h:11:15: note: forward declaration of 'class esphome::aux_airconditioner::AirCon'
         class AirCon;
               ^~~~~~
src/esphome/components/aux_ac/automation.h: In member function 'void esphome::aux_airconditioner::AirConPowerLimitationOffAction<Ts>::play(Ts ...)':
src/esphome/components/aux_ac/automation.h:146:52: warning: invalid use of incomplete type 'class esphome::aux_airconditioner::AirCon'
             void play(Ts... x) override { this->ac_->action_power_limitation_off(); }
                                                    ^~
src/esphome/components/aux_ac/automation.h:11:15: note: forward declaration of 'class esphome::aux_airconditioner::AirCon'
         class AirCon;
               ^~~~~~
src/esphome/components/aux_ac/automation.h: In member function 'void esphome::aux_airconditioner::AirConPowerLimitationOnAction<Ts>::play(Ts ...)':
src/esphome/components/aux_ac/automation.h:161:26: warning: invalid use of incomplete type 'class esphome::aux_airconditioner::AirCon'
                 this->ac_->action_power_limitation_on(this->value_.value(x...));
                          ^~
src/esphome/components/aux_ac/automation.h:11:15: note: forward declaration of 'class esphome::aux_airconditioner::AirCon'
         class AirCon;

GrKoR, 2024-03-05:
Hi!

I know about these warnings. They aren't affect anything.
They will be fixed. Someday =)

akdri, 2024-03-05:
I updated the devices to the new version.

All devices become unavailable after 5 minutes. I have to remove power from them to restart them, but every 5 minutes they lock and lose connection.

My code is the following:

external_components:
  - source: github://GrKoR/esphome_aux_ac_component
    components: [ aux_ac ]
    refresh: 0s

substitutions:
  devicename: aire_despacho
  upper_devicename: Aire Despacho

esphome:
  name: $devicename
  friendly_name: ${upper_devicename}                     

esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino

# Enable logging
logger:
  #  hardware_uart: USB_SERIAL_JTAG
    # set hardware_uart to UART1 and comment out baud_rate above in case of boot crashes
    # it is suitable if you need hardware loggin
  #hardware_uart: UART1
  level: DEBUG
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"

ota:
  password: "XXXXXXXXXXXXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.72.43
    gateway: 192.168.72.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Aire-Despacho AP"
    password: !secret wifi_password

captive_portal:
    
# UART0 configuration for AUX air conditioner communication
uart:
  id: ac_uart_bus
  # ATTENTION! Use GPIO4 (D2) and GPIO5 (D1) as the TX and RX for NodeMCU-like boards!
  tx_pin: GPIO3
  rx_pin: GPIO1
  baud_rate: 4800
  data_bits: 8
  parity: EVEN
  stop_bits: 1

climate:
  - platform: aux_ac
    name: ${upper_devicename}
[... 178 lines cut ...]
    icon: "mdi:circle-small"
    mode: "slider"
    min_value: 0
    max_value: 6
    step: 1
    update_interval: 2s
    lambda: |-
      return id(${devicename}_vlouver_state).state;
    set_action:
      then:
        - aux_ac.vlouver_set:
            id: aux_id
            position: !lambda "return x;"

  - platform: template
    name: ${upper_devicename} Power Limit
    id: ${devicename}_power_limit
    icon: "mdi:battery-unknown"
    mode: "slider"
    min_value: 30
    max_value: 100
    step: 1
    set_action:
      then:
        - lambda: !lambda "id(aux_id).action_power_limitation_on( x );"

closed 2024-11-03 by GrKoR