GrKoR/esphome_aux_ac_component #101: Does not work with esp32

akdri, 2024-03-06:
I have 3 esp32 connected to the air conditioners, until now they had worked perfectly, but since I updated and made the new modifications, the esp32 end up disconnecting.

The esp32s never come back when they are disconnected, but an esp32-c3 that I have if it manages to recover the connection is very unstable and continually restarts.

ESP32 UPTIME
[image]

ESP32-C3 UPTIME

[image]

Firmware: 2024.2.2  ESPHOME
My Code.

external_components:
  - source: github://GrKoR/esphome_aux_ac_component
    components: [ aux_ac ]
    refresh: 0s

substitutions:
  devicename: aire_despacho
  upper_devicename: Aire Despacho

esphome:
  name: $devicename
  friendly_name: ${upper_devicename}                     

esp32:
  board: wemos_d1_mini32
  framework:
    type: arduino

# Enable logging
logger:
  #  hardware_uart: USB_SERIAL_JTAG
    # set hardware_uart to UART1 and comment out baud_rate above in case of boot crashes
    # it is suitable if you need hardware loggin
  #hardware_uart: UART1
  level: DEBUG
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: "j4325243542354235gbertb4hg54g54543g435fg="

ota:
  password: "47802e8223452438b1301b23316234543259be87f19d82a"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.72.43
    gateway: 192.168.72.1
    subnet: 255.255.255.0
  reboot_timeout: 0s

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Aire-Despacho AP"
    password: !secret wifi_password

captive_portal:
    
# UART0 configuration for AUX air conditioner communication
uart:
  id: ac_uart_bus
  # ATTENTION! Use GPIO4 (D2) and GPIO5 (D1) as the TX and RX for NodeMCU-like boards!
  tx_pin: GPIO3
  rx_pin: GPIO1
  baud_rate: 4800
  data_bits: 8
  parity: EVEN
  stop_bits: 1

climate:
  - platform: aux_ac
    name: ${upper_devicename}
    id: aux_id
    uart_id: ac_uart_bus
    period: 7s
    display_inverted: false
    optimistic: true
    display_state:
      name: ${upper_devicename} Display State
      id: ${devicename}_display_state
      internal: false
    indoor_ambient_temperature:
      name: ${upper_devicename} Indoor Ambient Temperature
      id: ${devicename}_indoor_ambient_temp
      internal: false
    indoor_coil_temperature:
      name: ${upper_devicename} Indoor Coil Temperature
      id: ${devicename}_indoor_coil_temp
      internal: false
    outdoor_ambient_temperature:
      name: ${upper_devicename} Outdoor Ambient Temperature
      id: ${devicename}_outdoor_ambient_temp
      internal: false
    outdoor_condenser_temperature:
      name: ${upper_devicename} Outdoor Condenser Temperature
      id: ${devicename}_outdoor_condenser_temp
      internal: false
    compressor_suction_temperature:
      name: ${upper_devicename} Compressor Suction Temperature
      id: ${devicename}_compressor_suction_temp
      internal: false
    compressor_discharge_temperature:
      name: ${upper_devicename} Compressor Discharge Temperature
      id: ${devicename}_compressor_discharge_temp
      internal: false
    defrost_temperature:
      name: ${upper_devicename} Defrost Temperature
      id: ${devicename}_defrost_temp
      internal: false
    defrost_state:
      name: ${upper_devicename} Defrost State
      id: ${devicename}_defrost_state
      internal: false
    inverter_power:
      name: ${upper_devicename} Inverter Power
      id: ${devicename}_inverter_power
      internal: false
    preset_reporter:
      name: ${upper_devicename} Preset Reporter
      id: ${devicename}_preset_reporter
      internal: false
    vlouver_state:
      name: ${upper_devicename} VLouvers State
      id: ${devicename}_vlouver_state
      internal: false
    visual:
      min_temperature: 16
      max_temperature: 32
      temperature_step:
        target_temperature: 0.5
        current_temperature: 0.1
    supported_modes:
      - HEAT_COOL
      - COOL
      - HEAT
      - DRY
      - FAN_ONLY
    custom_fan_modes:
      - MUTE
      - TURBO
    supported_presets:
      - SLEEP
    custom_presets:
      - CLEAN
      - HEALTH
      - ANTIFUNGUS
    supported_swing_modes:
      - VERTICAL
      - HORIZONTAL
      - BOTH

sensor:
  # just wifi signal strength for debug purpose only
  - platform: wifi_signal
    name: ${upper_devicename} WiFi Signal
    update_interval: 30s
    unit_of_measurement: "dBa"
    accuracy_decimals: 0
  - platform: uptime
    name: ${upper_devicename} Uptime Sensor

switch:
  - platform: template
    name: ${upper_devicename} Display
    lambda: |-
      if (id(${devicename}_display_state).state) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      - aux_ac.display_on: aux_id
    turn_off_action:
      - aux_ac.display_off: aux_id

  - platform: template
    name: ${upper_devicename} Power Limitation
    lambda: |-
      if (id(${devicename}_inverter_power).state) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      #- aux_ac.power_limit_on: aux_id
      - aux_ac.power_limit_on:
          id: aux_id
          limit: 40
    turn_off_action:
      - aux_ac.power_limit_off: aux_id

button:

  - platform: template
    name: ${upper_devicename} VLouver Stop
    icon: "mdi:circle-small"
    on_press:
      - aux_ac.vlouver_stop: aux_id

  - platform: template
    name: ${upper_devicename} VLouver Swing
    icon: "mdi:pan-vertical"
    on_press:
      - aux_ac.vlouver_swing: aux_id

  - platform: template
    name: ${upper_devicename} VLouver Top
    icon: "mdi:pan-up"
    on_press:
      - aux_ac.vlouver_top: aux_id

  - platform: template
    name: ${upper_devicename} VLouver Middle Above
    icon: "mdi:pan-top-left"
    on_press:
      - aux_ac.vlouver_middle_above: aux_id

  - platform: template
    name: ${upper_devicename} VLouver Middle
    icon: "mdi:pan-left"
    on_press:
      - aux_ac.vlouver_middle: aux_id

  - platform: template
    name: ${upper_devicename} VLouver Middle Below
    icon: "mdi:pan-bottom-left"
    on_press:
      - aux_ac.vlouver_middle_below: aux_id

  - platform: template
    name: ${upper_devicename} VLouver Bottom
    icon: "mdi:pan-down"
    on_press:
      - aux_ac.vlouver_bottom: aux_id

number:
  - platform: template
    name: ${upper_devicename} Vertical Louvers
    id: ${devicename}_vlouver
    icon: "mdi:circle-small"
    mode: "slider"
    min_value: 0
    max_value: 6
    step: 1
    update_interval: 2s
    lambda: |-
      return id(${devicename}_vlouver_state).state;
    set_action:
      then:
        - aux_ac.vlouver_set:
            id: aux_id
            position: !lambda "return x;"

  - platform: template
    name: ${upper_devicename} Power Limit
    id: ${devicename}_power_limit
    icon: "mdi:battery-unknown"
    mode: "slider"
    min_value: 30
    max_value: 100
    step: 1
    set_action:
      then:
        - lambda: !lambda "id(aux_id).action_power_limitation_on( x );"

Is there a way to revert to the previous version?
because right now I can't control the air

thanks

GrKoR, 2024-03-06:
@akdri thank you for your feedback.

Yes, you can use previous versions of the component. Modify your config this way:

```yaml
external_components:
  - source:
      type: git
      url: https://github.com/GrKoR/esphome_aux_ac_component
      ref: v.0.2.14
    components: [ aux_ac ]
    refresh: 0s
```

akdri, 2024-03-06:
Thanks for your response, GrKoR

I have gone back to the previous version and now everything works correctly again.

As I also have several esp12e

I wanted to compile esp12e with your example to see if it is stable on the esp8266 platform.

But it has exactly the same problem, esl esp8266 is able to recover but it continually restarts approximately every 12 minutes.
[image]

GrKop thanks for your work, I hope you can make the new version stable or find the problem, for now I will continue with the old version that works perfectly, with fewer parameters but works well.

  In Spain, Baxi Anori air conditioners are brands of the AUX manufacturer. All of these air conditioners work with your protocol in case you want to add them to the list.

GrKoR, 2024-03-06:
@akdri it's very strange. All my air conditioners are working with esp12e based dongles. And they work stable...
I have no idea about the reason of this issue yet 🤔 

>

closed 2024-11-03 by GrKoR