GrKoR/esphome_aux_ac_component #102: Compressor Suction Temperature does not send negative values

akdri, 2024-03-06:
[image]

tomadam123, 2024-05-20:
The inability of a compressor suction temperature sensor to send negative values can stem from a variety of issues ranging from sensor limitations and calibration errors to system operating conditions and controller configurations. By systematically diagnosing each potential cause and applying the appropriate solutions  [useful source](https://plumbingheatingcoolingnj.com/)

GrKoR, 2024-05-21:
The @tomadam123 comment was hidden.
Reasons:
* it looks like spam (strange link in the message);
* author doesn't reply for my question in the issue #97 but posted another one comment with similar content here.

closed 2024-11-03 by GrKoR