GrKoR/esphome_aux_ac_component #105: cant update or install to new esp32

vietxtlife, 2024-03-11:
below is the error log, any ideas? thank you

INFO ESPHome 2024.2.2
INFO Reading configuration /config/esphome/living-room-ac.yaml...
INFO Updating https://github.com/GrKoR/esphome_aux_ac_component.git@None
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1041, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1019, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 1066, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 920, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 908, in _load_config
    result = validate_config(config, command_line_substitutions)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 836, in validate_config
    result.run_validation_steps()
  File "/esphome/esphome/config.py", line 143, in run_validation_steps
    task.step.run(self)
  File "/esphome/esphome/config.py", line 317, in run
    component = get_component(self.domain)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/loader.py", line 184, in get_component
    assert "." not in domain
           ^^^^^^^^^^^^^^^^^
AssertionError

GrKoR, 2024-03-11:
Hi!

Show me your config please (external_components part).

vietxtlife, 2024-03-11:
where does that config file located, i cant see where it at. in home assistant right?

GrKoR, 2024-03-11:
As I can see from your log, it should be in /config/esphome/living-room-ac.yaml

vietxtlife, 2024-03-14:
external_components:
  - source: github://GrKoR/esphome_aux_ac_component
    components: [ aux_ac ]
    refresh: 0s

esphome:
  name: living-room-ac
  platform: esp32
  board: esp32dev

wifi:
  ssid: daV
  password: !secret
  #manual_ip:
    #static_ip: 
    #gateway: 
    #subnet: 
  ap:
    ssid: living-room-ac Hotspot
    password: !secret ap_password
#  use_address: ${wifi_ota_ip}

captive_portal:
debug:

logger:
  #  level: DEBUG
    baud_rate: 0
    # set hardware_uart to UART1 and comment out baud_rate above in case of boot crashes
    # it is suitable if you need hardware loggin
    hardware_uart: UART1

api:
  encryption:
    key: "eVzsIH123RBPje+aV1KIqeSenju0XZcZwtsFSoH4L1I="

ota:
  password: "ab018d547b945c70b5d39123dd6cddad"

# UART0 configuration for AUX air conditioner communication
uart:
  id: ac_uart_bus
  # ATTENTION! Use GPIO4 (D2) and GPIO5 (D1) as the TX and RX for NodeMCU-like boards!
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 4800
  data_bits: 8
  parity: EVEN
  stop_bits: 1

climate.aux_ac:
  - platform: aux_ac
    name: "living-room-ac"
    id: aux_id
    uart_id: ac_uart_bus
    period: 7s

    display_inverted: false
    timeout: 301
    optimistic: true
    indoor_ambient_temperature:
      name: living-room-ac Indoor Ambient Temperature
      id: aux_id_indoor_ambient_temp
      internal: false
    indoor_coil_temperature:
      name: living-room-ac Indoor Coil Temperature
      id: aux_id_indoor_coil_temp
      internal: false
    outdoor_ambient_temperature:
      name: living-room-ac Outdoor Ambient Temperature
      id: aux_id_outdoor_ambient_temp
      internal: false
    outdoor_condenser_temperature:
      name: living-room-ac Outdoor Condenser Temperature
      id: aux_id_outdoor_condenser_temp
    compressor_discharge_temperature:
      name: AC Compressor Temperature
      id: ac_strange_temp
      internal: false
    display_state:
      name: AC Display State
      id: ac_display_state
      internal: false
    defrost_state:
      name: AC Defrost State
      id: ac_defrost_state
      internal: false
    inverter_power:
      name: AC Inverter Power
      id: ac_inverter_power
      internal: false
    inverter_power_limit_value:
      name: AC Inverter Power Limit Value
      id: ac_inverter_power_limit_value
      internal: false
    inverter_power_limit_state:
      name: AC Inverter Power Limit State
      id: ac_inverter_power_limit_state
      internal: false
    preset_reporter:
      name: AC Preset Reporter
      id: ac_preset_reporter
      internal: false
    vlouver_state:
      name: AC Vertical Louvers State
      id: ac_vlouver_state
      internal: false
    visual:
      min_temperature: 16
      max_temperature: 32
      temperature_step: 1
    supported_modes:
      - HEAT_COOL
      - COOL
      - HEAT
      - DRY
      - FAN_ONLY
    custom_fan_modes:
      - MUTE
      - TURBO
    supported_presets:
      - SLEEP
    custom_presets:
      - CLEAN
      - HEALTH
      - ANTIFUNGUS
    supported_swing_modes:
      - VERTICAL
      - HORIZONTAL
      - BOTH

GrKoR, 2024-03-14:
Hello!
You forgot to change one of substitutions with specific value. Change it for 'use_address' parameter.

closed 2024-03-14 by GrKoR

vietxtlife, 2024-03-14:
Hey, what should I change it to? Thank you

GrKoR, 2024-03-14:
Sorry. There was no correct formatting of your config and I decided that you use parameter 'use_address'. But I think that esphome parsing substitutions even when they are commented. Delete string with 'use_address'. Probably it fixes the error.
And don't forget to set wifi password. Your config has this parameter with '!secret' keyword but without password variable.

vietxtlife, 2024-03-14:
Thank you, i will try that

vietxtlife, 2024-03-14:
INFO ESPHome 2024.2.2 INFO Reading configuration
/config/esphome/living-room-ac.yaml... INFO Updating
***@***.*** ERROR Unexpected
exception while reading configuration: Traceback (most recent call last): File
"/usr/local/bin/esphome", line 33, in <module>
sys.exit(load_entry_point('esphome',
'console_scripts', 'esphome')())
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/esphome/esphome/__main__.py", line 1041, in main return
run_esphome(sys.argv) ^^^^^^^^^^^^^^^^^^^^^ File
"/esphome/esphome/__main__.py", line 1019, in run_esphome config =
read_config(dict(args.substitution) if args.substitution else {})
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/esphome/esphome/config.py", line 1066, in read_config res =
load_config(command_line_substitutions) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/esphome/esphome/config.py", line 920, in load_config return
_load_config(command_line_substitutions)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/esphome/esphome/config.py", line 908, in _load_config result =
validate_config(config, command_line_substitutions)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/esphome/esphome/config.py", line 836, in validate_config
result.run_validation_steps()
File "/esphome/esphome/config.py", line 143, in run_validation_steps
task.step.run(self)
File "/esphome/esphome/config.py", line 317, in run component =
get_component(self.domain) ^^^^^^^^^^^^^^^^^^^^^^^^^^ File
"/esphome/esphome/loader.py", line 184, in get_component assert "." not in
domain ^^^^^^^^^^^^^^^^^ AssertionError

New errors, please help. Thank you so much for your work

GrKoR, 2024-03-14:
This isn't a component error. You did something wrong with config but I don't know what is exactly wrong. Try to get simple example config from GitHub and add features to it step by step.

vietxtlife, 2024-03-14:
Ok, I will do that. Thank you