GrKoR/esphome_aux_ac_component #110: AUX ASW-H12C5C4/JER3DI-B8-2 is working good!

tomycz, 2024-05-02:
Thank you very much for your development, 🤝 I made your project succesfully with USB connector! :)
It's look like working well, and all function is good.  

My climate is: **AUX ASW-H12C5C4/JER3DI-B8-2**
The original AC USB stick number is: **CTTG-40X24-WIFI-AKS2, 2022 04 18**

Please write it to the supported AC list. Thanks!

GrKoR, 2024-05-03:
Hey!
Thank you for your feedback. I'll add your AC to the list soon.

kbalint, 2024-05-31:
@tomycz I have nearly the same Delta 2 (7.2). Could I ask you which version did you do? the USB connector output with +8.5V? thanks! Did you try any other integrations too? I've tried the homebridge/mqtt versions, but neither of them worked (couldn't even find the AC on the network) I guess the AKS[2] stick is the problem.

tomycz, 2024-05-31:
I'm hungarian too. 😃
You can find me PM, or email (mailto: tomycz86@gmail.com), but i attached some pics from my device datasheet:

[image][image]

closed 2024-06-19 by GrKoR