GrKoR/esphome_aux_ac_component #112: inverter power limitation: no feedback for device's Power Limit slider + relative / absolute limit

GrKoR, 2024-05-24:
applicable to component v.0.2.15

Brokly & Alexander Mamonov in telegram:
> Привет всем, есть идеи, почему может не включаться лимит по питанию?
> В чем выражается не установка лимита ?
> не сохраняется ползунок. ошибка в логе [выше](https://t.me/aux_ac/20136)
> а похоже, просто ползунок не обновляет свое состояние
> dr_corner , похоже у тебя в примере конфига на гитхабе есть косяк. Нет обратной связи с ползунком установки мощности.
> но и кондей забивает на лимит
> Нет , по логу не забивает.
> Ставит столько, на сколько установлен ползунок
> да, лимит ставится, но кондей уходит за лимит
> Нет, просто для кондея 100% соответствует установленному лимиту. Это разные проценты. Скажем ставишь 71 процент, кондей показывает, что работает на 40, это значит он работает на 0.71*0.4=0.28, то есть на 28 процентов
> Кстати это тоже можно обслужить в конфиге.
> Если хочется видеть не относительные показания процентов, а абсолютные.

Correct config [here](https://t.me/aux_ac/20168)

Steamerzone, 2024-12-04:
Can you post the correct config here on github?

Thanks.

GrKoR, 2024-12-06:
Hi, @Steamerzone !

Here is the contents of the config file from telegram you were asking for. Note, that it is common part of [the advanced example](https://github.com/GrKoR/esphome_aux_ac_component/tree/master/examples/advanced).

```yaml
# DON'T COMPILE THIS FILE
# This file contains common settings for all air conditioners of your house
external_components:
  - source: github://GrKoR/esphome_aux_ac_component
    components: [ aux_ac ]
    refresh: 0s

esphome:
  name: $devicename
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "****************************"
  password: "************************"
  manual_ip:
    static_ip: ${wifi_ip}
    gateway: 192.168.83.1
    subnet: 255.255.255.0
  ap:
    ssid: ${upper_devicename} Hotspot
    password: "*********************************"

captive_portal:
debug:

logger:
    level: DEBUG
    baud_rate: 0
    # set hardware_uart to UART1 and comment out baud_rate above in case of boot crashes
    # it is suitable if you need hardware loggin
    # hardware_uart: UART1

api:
  password: "********************"

ota:
  password: "*****************************"

# UART0 configuration for AUX air conditioner communication
uart:
  id: ac_uart_bus
  # ATTENTION! Use GPIO4 (D2) and GPIO5 (D1) as the TX and RX for NodeMCU-like boards!
  tx_pin: GPIO5
  rx_pin: GPIO4
  baud_rate: 4800
  data_bits: 8
  parity: EVEN
  stop_bits: 1

climate:
  - platform: aux_ac
    name: ${upper_devicename}
    id: aux_id
    uart_id: ac_uart_bus
    period: 7s
    show_action: true
    display_inverted: false
    optimistic: false
    indoor_temperature:
      name: ${upper_devicename} Indoor Temperature
      id: ${devicename}_indoor_temp
      internal: false
    display_state:
      name: ${upper_devicename} Display State
[... 145 lines cut ...]
    min_value: 0
    max_value: 6
    step: 1
    update_interval: 2s
    lambda: |-
      return id(${devicename}_vlouver_state).state;
    set_action:
      then:
        - aux_ac.vlouver_set:
            id: aux_id
            position: !lambda "return x;"
  - platform: template
    name: ${upper_devicename} Power Limit
    id: ${devicename}_power_limit
    icon: "mdi:battery-unknown"
    mode: "slider"
    min_value: 30
    max_value: 100
    step: 1
    set_action:
      then:
        - lambda: !lambda |-
            id(aux_id).powerLimitationOnSequence( x );
            id(${devicename}_power_limit).publish_state( x );
```