GrKoR/esphome_aux_ac_component #113: IKON air conditioner

thepuncake, 2024-06-10:
Hi there, I have an IKON ac as the title states which looks like it could be aux made. It has a female USB port inside that on the other side is connected to the board/display of the ac. This is written on the back of the board/display:

11222014000650
CTTK-99X57-AKS
2021-11-02

I'm including two pictures, one of the ac with the cover raised and the sticker on the outside unit. I couldn't find any info on this online, seems like this model is only available in Cyprus.
[image]
[image]

Would appreciate any feedback. I'm hoping that the pseudo USB connector in the documentation has the same pinout as my unit.

thepuncake, 2024-06-11:
Thanks, it works! I'm using a lolin s2 mini with a bidirectional level shifter like so:
[image]
Great work on the component! I guess u can go ahead and add my AC to the list of supported devices.

GrKoR, 2024-06-11:
Congrats!)
I'll add your AC to the list of tested units.

closed 2024-06-19 by GrKoR