GrKoR/esphome_aux_ac_component #117: New AC tested: Rinnai RINV25RC

BenJamesAndo, 2024-07-03:
Thanks for this great project. I've successfully got it controlling my Rinnai AC. Model `RINV25RC`
It uses the 5-wire connection.
[image]
[image]

Parts list
[ESP-32 Dev Board 38 Pins](https://www.ebay.com.au/itm/386377067231)
[5 Pin JST SM Cable](https://www.ebay.com.au/itm/153937969850?var=454006808168)
[ESP32 Breakout Board](https://www.makerstore.com.au/product/elec-esp32-devc-msbob/). I got this one as it could take in 12V via terminal blocks and step it down to 3.3V for the ESP32.
I might redo it with smaller & cheaper parts to free up these parts for other projects.

GrKoR, 2024-07-03:
Hey! Thank you for your feedback! =) I'll add your AC to the tested list soon.

GrKoR, 2024-07-13:
added

closed 2024-07-13 by GrKoR