GrKoR/esphome_aux_ac_component #121: New AC tested: Baymak Elegant Plus 9

cempehlivan, 2024-07-14:
Hello, thank you very much for your great project! It worked perfectly for my Baymak Elegant Plus 9 air conditioner.

[image]

[image]

GrKoR, 2024-07-15:
@cempehlivan Thank you for the feedback! Will add it to the list.

P.S. Hope, you will redo wires connection after first tests :)
I have never seen using of crimping tips for pins connection. It's awesome!

cempehlivan, 2024-07-15:
Of course these pin connections were for testing :) Now I made up a box and finalized it.

[image]
[image]
[image]

earvingad, 2024-10-26:
@cempehlivan what voltage regulator and capacitor are you using?
Sorry to bother you.

closed 2025-02-21 by GrKoR