GrKoR/esphome_aux_ac_component #122: Became unavailable

cempehlivan, 2024-08-02:
Hello. Do you have any idea about this error? I searched for the problem myself but couldn't find any problem.

[image]

GrKoR, 2024-08-02:
Hi!

Some users are reported about the same behavior. And we can't figure out what is the reason of reboots yet.
I have in use four of AC with my component. And only one is doing the same.

Our assumption that it is most likely a hardware issue. Some kind of troubles with power supply or with 5V logic level of AC's signals. But this is just a guesses.
Now we are experimenting with capacitors on the VCC pins of ESP and with level shifters on the TX and RX lines. But we haven't results yet because this issue requires the accumulation of statistics.

If you are familiar with electronics you can add level shifter to your device and take a part in the statistics gathering.

cempehlivan, 2024-08-05:
I tried it with L7805 voltage regulator and 10 mu capacitor. Then I tried it with XL4005 DC/DC converter. The result did not change.

cempehlivan, 2024-08-05:
When the air conditioner went to "Became unavailable" status, I checked from the router and saw that the device was not connected to the wifi network. Although the device is close to the router, it does not appear to be connected. I have another esp8266 device (not an air conditioner) at a further distance from the router and there is no problem with it.

GrKoR, 2024-08-05:
Yes, that's all the same as other men reported.

Device became unavailable cause it is rebooting. The reason of reboots isn't clear yet. Crash reports from esp core gave us nothing.

Because you tested power supply assumption, there's one more possible way for tests: logic level shifter.

Why do we think that this issue has its roots in the hardware? Because I buy a esp8266 based dongle designed by a man from community. This dongle is a small factory produced pcb with esp12, step down dc-dc and logic level shifter on the board. I had reboots with my DIY esp12e on the one of my air conditioners. But when I changed it with dongle from community reboots are gone.
The difference between this two devices is factory produced pcb, good electrical design, good power and level shifters.

That's all what we have around this issue :)

cempehlivan, 2024-08-07:
I added a logic level shifter. There is no problem for now, it works. I will follow its operation for a while and write the result again.

[image]

GrKoR, 2024-08-08:
Wow! It's great!
Thank you for your experiments!

cempehlivan, 2024-08-08:
Unfortunately that didn't work either. Is there a diagram of the PCB shared in the community?

[image]

GrKoR, 2024-08-08:
Did you implement uptime sensor in the config? It is a little bit more indicative than home assistant log.

You can find hardware design here: https://github.com/dudanov/iot-uni-dongle
This isn't a last version of the dongle but all design decisions are shown correctly.

cempehlivan, 2024-08-12:
Let me share the changes I have made in the last few days and their results;
I replaced my D1 mini v3 card with a D1 mini v4.
I replaced the logic level shifter with a new one.
I soldered a new pertinax as cleanly as possible.
I assigned a static IP address to the device.
I use a mesh system in my home and I limited the roaming of this device in the mesh system and fixed it to a single route and configured it to use only the 2.4 Ghz band.
As a result, it has been reset once in the last 2 days. I am still following it. I added detailed analysis sensors. I will check if it resets :)

[image]

[image]

GrKoR, 2024-08-14:
Cool investigation! 👍🏼
It's very sad that it still reboots. Two days isn't what we expected.

As I know another one option is to migrate from esp8266 to esp32.

rulgzz, 2024-09-21:
> When the air conditioner went to "Became unavailable" status, I checked from the router and saw that the device was not connected to the wifi network. Although the device is close to the router, it [...]

I have the same issue with the same ESP8266 D1 mini, it works fine for about 30 minutes, then goes offline, ping start to get higher response times then no response at all.
It last longer when the A/C is off.
Also using logic level shifter and step down regulator.
A/C is AUFIT (rebranded AUX)

pgpanayotov, 2024-11-21:
Hi, I have similar issue with Arielli AC (rebranded AUX).

Using D1 mini and dc-dc power convertor and plain TX/RX connections. I have level shifters, but what I can see from above that might not solve the problem.
No issues reported by other D1 Mini's that are far away from my router. So I think that router/connectivity setup is all good.

The ESP unit rebooted couple of times for the last 24h. 
What I done so far is to move it to another AC unit (I have 4 with different sizes 9000, 12000 and 18000 BTU).
Haven't seen a reboot since I moved it, but if it keeps doing it, maybe I can try to power it externally.

I wonder if anyone tried to power the ESP externally ? 
I'm powering couple of my ESP's with small buck 220 to 5V PSU's.

GrKoR, 2024-11-21:
As guys reports in the telegram, there are no reboots with ESP32 C3. So I recommend this chip for further installations.
Most likely sporadic reboots are the result of hardware bugs of esp8266.