GrKoR/esphome_aux_ac_component #124: new ac tested and working. Costway (AUX) FP10524US-22WH & FP10318US-22WH

albyjr, 2024-08-09:
Costway rebadged AUX unit in the United States. 
The FP10524US came with wifi the newer wifi stick CTTG-40x24-Wifi-AKS2. Everything working as it's supposed to. 

The FP10318US-22H did not come with wifi but on the ciruit board it had the holes for the connector but no pin header. 
I added my own pin header and rigged it up the same way as the 24k unit and all is working fine. 
[image]
[image]
[image]
[image]
[image]

GrKoR, 2024-08-10:
Cool!
Thank you for this info.

closed 2025-02-21 by GrKoR

at012423, 2025-12-06:
> Interesting - I have a Garvee branded unit using this same adapter - trying to get it working - and noticing it's only sending 7.5v to the adapter. It shows 12v in your photos - so wondering if that's the issue with mine.
> <img alt="Image" width="929" height="1238" src="https://private-user-images.githubusercontent.com/13157645/520409670-2b8169da-c805-41d9-8675-a2cfb256ce1a.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjUwMzAzMzQsIm5iZiI6MTc2NTAzMDAzNCwicGF0aCI6Ii8xMzE1NzY0NS81MjA0MDk2NzAtMmI4MTY5ZGEtYzgwNS00MWQ5LTg2NzUtYTJjZmIyNTZjZTFhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTEyMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUxMjA2VDE0MDcxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYyY2JhMjhlYzlmMGY5MWMyOWFmYzgyMWM4NmQ4YTgyMzEwMGI2ZDVmOTU1NzZlNmI1NWFlNThhNTFhODMyMjYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.oeS7xdfdW24VIpZr-ayGl7jOKvQBIAsrLoGApxFEyOw">

I believe this is due to your tester only supporting up to 7v, erroring out at the the 7.49 value. Having similar connectivity issues with mine however, stumbled into this thread troubleshooting.