GrKoR/esphome_aux_ac_component #125: Power consumption calculation

drupry, 2024-08-14:
Hi, this is not an issue but i don't know how to do it. I want to measure at least approximate power consumption but I don't have extra space in my circuit breaker box for some Shelly relay or something so I decided to calculate power consumption from Inverter power entity. 
My cooling power input is 1340W and heating power input is 950W. I have crated a sensor template which is calculating inverter power * 1340W but obviously it will work only in summer when the cooling is on. Is there any way how to automate this ? how the sensor template know when the AC is set for cooling and when for heating ? Because i want to include HEAT_COOL mode also. Or is this sooo much inaccurate that I should get out of it ? 

This is how my sensor template looks like now

```- platform: template
     sensors:
     power_ac_bedroom:
     friendly_name: "Electric power ac bedroom"
     unit_of_measurement: 'W'
     value_template: "{{ states('sensor.ac_bedroom_ac_inverter_power') | float / 100 * 1340 }}"

GrKoR, 2024-08-14:
Hi!

This kind of calculation is dramatically inaccurate. It depends on Mars weather and Moon phase 😀 

You can try to place your shelly or another measuring device here:
[image]

GrKoR, 2024-08-14:
BTW, if you want to experiment with it anyway, then use 'action':
'''C++
// Current action (currentl on idle, cooling, heating, etc.), ClimateAction (enum)
id(my_climate).action
'''

Brokly, 2026-09-22:
[image]

GrKoR, 2026-09-22:
Выглядит так, что можно пересчитать по коэффициенту.
Но мне не актуально)