GrKoR/esphome_aux_ac_component #128: Support for esp-idf framework?

rvalitov, 2024-10-08:
Hi! I'm using ESP32-C3 super mini with the following config:
```yaml
esp32:
  board: esp32-c3-devkitm-1
  variant: ESP32C3
  framework:
    type: esp-idf
```

This leads to compilation error:
```
In file included from src/esphome\components\aux_ac\automation.h:3,
                 from src/esphome.h:15,
                 from src/main.cpp:3:
src/esphome\components\aux_ac\aux_ac.h:7:10: fatal error: Arduino.h: No such file or directory

*****************************************************************
* Looking for Arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Arduino.h"
* Web  > https://registry.platformio.org/search?q=header:Arduino.h
*
*****************************************************************

 #include <Arduino.h>
          ^~~~~~~~~~~
compilation terminated.
```

If I change the framework to `arduino` then the compilation works, but in all examples for this chip it's recommended to use `esp-idf`. So the question is `arduino` framework is really obligatory here or it's possible to make it work with `esp-idf` too?

GrKoR, 2024-10-08:
Hey! I don't even remember what part of Arduino framework did I use in the component. Most likely I can make the code independent from the framework. But it will not be soon cause I haven’t enough time yet.

GrKoR, 2024-10-25:
I don't know the way, because I haven't enough time for digging deeper with it yet. If you are familiar with C/C++ and esp-idf, it would be very appreciated if you create pull request with such code changes.