GrKoR/esphome_aux_ac_component #129: 4-pin connection

rvalitov, 2024-10-08:
Hi! I have Energolux Champery with a 4-pin connection:
- red
- black
- orange
- brown

Any idea which color to use? In the docs I found only info for 5-pin connection.
Photos:
[image]
[image]
[image]

GrKoR, 2024-10-08:
Hi!
Probably it isn't an AUX-based air conditioner. Did you check it?

rvalitov, 2024-10-08:
Thank you so much for a prompt response! Well, a good question. Now you made me doubt it, so I checked some more, let me share my investigation. 

- The model is [Energolux CHAMPERY SAS07CH1-AI](https://energolux.shop/sas07ch1-a). 
- The supported WiFi adapter according to the manufacturer is [Energolux SIW7.9.12G1 Wi-Fi](https://energolux.shop/energolux-siw7912g1). 
- According to the [manual](https://www.severcon.ru/upload/iblock/cc2/v9wvf9yyh2yir550kn3ajumnqo38ypfg/Instruktsiya-po-ekspluatatsii_ENERGOLUX_Champery.pdf) the app to install is [EWPE Smart](https://play.google.com/store/apps/details?id=com.gree.ewpesmart).

Does it mean that this AC is not AUX-based?

GrKoR, 2024-10-08:
Yes, most likely it isn't made by AUX. Additional, none of the component's users reported compatibility with this Energolux model...
 
You can find out the manufacturer of your AC in one of these ways:
* Find line "manufacturer" in the bottom of the label on the indoor and/or outdoor unit of your AC. There is real manufacturer data often placed here.
* Another way is to make google search with photo of your AC's IR-remote. Search engines are perfect in this task today.

rvalitov, 2024-10-09:
Thank you. Image googling says it's a Gree based AC. It means it won't work right? Perhaps you have any advice for me?

rvalitov, 2024-10-09:
Oh thank you so much! No problem with the language! Your help is very important!

closed 2024-10-09 by rvalitov