GrKoR/esphome_aux_ac_component #148: unavailable

arsenal3008, 2025-03-21:
yesterday after the update I received this message in the log:
[14:11:58][W][AirCon:2091]: Sequence [step 1]: irrelevant incoming packet
[14:11:58][W][AirCon:2092]: Incoming packet:
[14:11:58][W][AirCon:2093]: 0000287686: [<=] [BB 00 07 00 00 00 18 00] 01 2C C0 20 00 00 00 3A 00 3A 00 00 00 00 00 00 00 00 00 00 00 00 00 03 [64 3B] 
[14:11:58][W][AirCon:2094]: Sequence packet needed: PACKET_TYPE = 07, CMD = 11
[14:11:58][W][AirCon:1081]: Sequence  [step 1]: error was occur in step function
today the device is not detected

arsenal3008, 2025-03-21:
похоже сгорела еспишка, постонно горит красным и винда не видит

GrKoR, 2025-03-21:
Тогда закрываю ишью. Если вдруг окажется, что проблема не в еспишке, пиши тут в закрытой - я увижу

closed 2025-03-21 by GrKoR

arsenal3008, 2025-03-21:
ок