GrKoR/esphome_aux_ac_component #151: Ballu - bsai/in-12hn1_15Y

Romko666, 2025-04-19:
Привет, воспользовался инструкцией, залил на ESP32 код через HA - esphome builder.
Лог загрузки скода и старта платы во вложении.

Физически подключен кондиционер к плате к 41 и 42 пинам (менял местами провода, пробовал подключать к 27 и 25 пинам).

[image]

В HA при этом данные не отражаются (температуры и т.д.), при попытке из HA выполнить старт\стоп или ещё что ошибка:

> ```
> [17:44:31][D][climate:011]: 'Ballu' - Setting
> [17:44:31][D][climate:015]:   Mode: COOL
> [17:44:31][E][AirCon:3383]: commandSequence: no pings from HVAC. It seems like no AC connected.
> [17:44:31][D][climate:396]: 'Ballu' - Sending state:
> [17:44:31][D][climate:399]:   Mode: COOL
> [17:44:31][D][climate:401]:   Action: IDLE
> [17:44:31][D][climate:404]:   Fan Mode: LOW
> [17:44:31][D][climate:407]:   Custom Fan Mode: 
> [17:44:31][D][climate:410]:   Preset: NONE
> [17:44:31][D][climate:413]:   Custom Preset: 
> [17:44:31][D][climate:416]:   Swing Mode: OFF
> [17:44:31][D][climate:419]:   Current Temperature: nan°C
> [17:44:31][D][climate:425]:   Target Temperature: nan°C
> [17:44:31][V][sensor:043]: 'AC Indoor Temperature': Received new state 0.000000
> [17:44:31][D][sensor:094]: 'AC Indoor Temperature': Sending state 0.00000 °C with 1 decimals of accuracy
> [17:44:31][V][sensor:043]: 'AC Outdoor Temperature': Received new state 0.000000
> [17:44:31][D][sensor:094]: 'AC Outdoor Temperature': Sending state 0.00000 °C with 0 decimals of accuracy
> [17:44:31][V][sensor:043]: 'AC Coolant Inbound Temperature': Received new state 0.000000
> [17:44:31][D][sensor:094]: 'AC Coolant Inbound Temperature': Sending state 0.00000 °C with 0 decimals of accuracy
> [17:44:31][V][sensor:043]: 'AC Coolant Outbound Temperature': Received new state 0.000000
> [17:44:31][D][sensor:094]: 'AC Coolant Outbound Temperature': Sending state 0.00000 °C with 0 decimals of accuracy
> [17:44:31][V][sensor:043]: 'AC Compressor Temperature': Received new state 0.000000
> [17:44:31][D][sensor:094]: 'AC Compressor Temperature': Sending state 0.00000 °C with 0 decimals of accuracy
> [17:44:31][V][sensor:043]: 'AC Inverter Power': Received new state 0.000000
> [17:44:31][D][sensor:094]: 'AC Inverter Power': Sending state 0.00000 % with 0 decimals of accuracy
> [17:44:31][D][binary_sensor:036]: 'AC Defrost State': Sending state OFF
> [17:44:31][D][AirCon:3667]: AUXvlouverToVlouverFrontend: unknown vertical louver state = 255
> [17:44:31][V][sensor:043]: 'AC Vertical Louvers State': Received new state 1.000000
> [17:44:31][D][sensor:094]: 'AC Vertical Louvers State': Sending state 1.00000  with 0 decimals of accuracy
> [17:44:31][D][binary_sensor:036]: 'AC Inverter Power Limit State': Sending state OFF
> [17:44:31][V][sensor:043]: 'AC Inverter Power Limit Value': Received new state 255.000000
> [17:44:31][D][sensor:094]: 'AC Inverter Power Limit Value': Sending state 255.00000 % with 0 decimals of accuracy
> [17:44:31][V][text_sensor:013]: 'AC Preset Reporter': Received new state NONE
> [17:44:31][D][text_sensor:064]: 'AC Preset Reporter': Sending state 'NONE'
> ```

[logs_condicioner_run.txt](https://github.com/user-attachments/files/19821421/logs_condicioner_run.txt)
[condicioner.yaml.txt](https://github.com/user-attachments/files/19821422/condicioner.yaml.txt)

Подскажите, пожалуйста, в чем может быть ошибка, что я не так делаю?

GrKoR, 2025-04-19:
Привет!
А кондиционер точно AUX?
Ballu делают свои кондеи на разных заводах, не факт что у вас под капотом аукс. В протестированных такой модели не вижу

GrKoR, 2025-04-19:
Аукс - это завод-производитель. Судя по фотке платы у вас точно не Аукс. А если смотреть по фото шильдика с моделью, у вас Грин (латвийский перекуп). Ауксы через них не приходили никогда.
Идите в чат в телеге, там в закрепе FAQ. В нем по шагам расписано, как определить производителя.

GrKoR, 2025-04-19:
Вдогонку: похоже, производитель TCL, но лучше проверить, как описано в FAQ

Romko666, 2025-04-19:
В чат меня не пустили сегодня.
Ок, скорее всего беда-печаль с данным кондиционером вопреки ожиданиям. Попробую IR.
Благодарю за ответы.

GrKoR, 2025-04-19:
В чате бот с капчей на входе. Если не ответить ему, то выпиливает. Пробуйте ещё и на этот раз не пропустите капчу)