GrKoR/esphome_aux_ac_component #165: AUX status displaying incorrect values

Steamerzone, 2025-06-17:
The AUX status is currently displaying incorrect values. When it's cooling, it shows Drying; when it's heating, it shows Idle; and when it's in Fan Only mode, it also shows Idle.

I'm not sure when this stopped working, but I had automations set up based on these values, and they were functioning correctly before.

It’s possible this change might be related to a recent ESPHome or Home Assistant update?

Steamerzone, 2025-06-17:
Looking through the code I see that it uses calculated values for the action:

i.e.:
                            int16_t delta_temp = _current_ac_state.temp_ambient - _current_ac_state.temp_inbound;
                            if (delta_temp > 0 && delta_temp < 2 &&
                                (_current_ac_state.realFanSpeed == AC_REAL_FAN_OFF ||
                                 _current_ac_state.realFanSpeed == AC_REAL_FAN_MUTE ||
                                 _current_ac_state.realFanSpeed == AC_REAL_FAN_MUTE))
                            {
                                this->action = climate::CLIMATE_ACTION_DRYING; // ОСУШЕНИЕ
                            }

closed 2025-06-17 by Steamerzone

GrKoR, 2025-06-17:
Some information about this behaviour is here: #118