GrKoR/esphome_aux_ac_component #178: AWX-12KTAI

stargate426, 2025-08-24:
I loaded an ESP32 with this firmware, when the device boots it sends some data to the AC but the AC never responds.

Is my AC (AWX-12KTAI) just not supported and can't work with this or am I doing something wrong?

I have wired GND <=> GND
ESP32 RX <=> (USB like AC WIFI connector middle pin 1)
ESP32 TX <=> (USB like AC WIFI connector middle pin 2)
[and swapped TX and RX]

used the following config:
```yaml
uart:
  id: ac_uart_bus
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 4800
  data_bits: 8
  parity: EVEN
  stop_bits: 1

climate:
  - platform: aux_ac
    name: "Room AC"
    uart_id: ac_uart_bus```

GrKoR, 2025-08-24:
Show me please the photos of label and remote

stargate426, 2025-08-24:
Here are images of the remote, label on the indoor unit, board where WiFi connector is located.

[image]
[image]
[image]

PS: if it matters I am using an ESP32-wroom-32D

closed 2025-08-24 by GrKoR